don/cordova-plugin-ble-central #106: Android scan not working properly

mathlarocque, 2015-10-10:
Hi, I'm currently trying to use your plugins with an android app, but when I ble.startScan, I don't discover any device. At first I thought I had no bluetooth device near me, but when I go to the setting under bluetooth, I can detect 2 device. I assume they should be listed when I ble.startScan... I got no error. I tried to pair the device with the core of android and then to scan to get the device but it doesn't work. Is there a procedure I must follow to ensure everything work ? 

I verified that the bluetooth is enabled with ble.isEnabled (which worked because it turn on the bluetooth on the device)

I'm using a nexus 4 with android 5.1.1

thank in advance for the help

morganRotaStabelli, 2015-10-12:
Sure near devices are BLE? Your phone can detect bluetooth end ble but that plugin is for BLE...
I suggest you get a scan application for ble 4.0 on PlayStore to see which communication protocol are using your devices

don, 2015-10-12:
You need BLE devices near your phone that are advertising. 

What specific device are you trying to discover?

 Use [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to verify there are available devices. Scanning using the Bluetooth Settings may be finding Bluetooth Classic devices. You shouldn't pair with most BLE devices.

The simplest code is to scan for any device and print the JSON out to the console. https://github.com/don/cordova-plugin-ble-central#quick-example

    ble.scan([], 5, function(device) {
        console.log(JSON.stringify(device));
    }, failure);

closed 2015-10-23 by don