don/cordova-plugin-ble-central #107: Plugin stopped working after clearing platforms/ios and rebuilding

AndreasHeiberg, 2015-10-13:
I'm lost as to how it used to work but now doesn't?

[Error](https://www.dropbox.com/s/4dvkwmlt9qisghe/Screenshot%202015-10-13%2015.32.29.png?dl=0)!

AndreasHeiberg, 2015-10-13:
Diff from when it worked 

https://www.dropbox.com/s/hpj0xekv107gsgk/Screenshot%202015-10-13%2015.42.15.png?dl=0

AndreasHeiberg, 2015-10-13:
Yeah I just reverted that file in git and now it works. I have no clue what the problem is, but seems like something that should be considered.

don, 2015-10-13:
The links to the screenshots don't work so I can't see what's happening. You can drag images into the text area when commenting on an issue.

AndreasHeiberg, 2015-10-15:
[image]
[image]
[image]

don, 2015-10-16:
Based on the diff it looks like you have both cordova-plugin-ble-central and cordova-plugin-bluetooth-serial installed. I don't think they'll work together on iOS.

closed 2015-10-21 by don