don/cordova-plugin-ble-central #108: Is there a limit to the number of devices returned from a call to scan?

halindrome, 2015-10-14:
I am in an environment where there are A LOT of BLE devices.  Sometimes I seem to get into a situation where the BLE device I am testing is advertising (I can see it from nRF Master) but when I "inspect" the app that is running on Android I never see my device.  It particularly seems to happen in a scenario like this:

App searches for and finds device
App connects to device
App gets notifications from device for a while
Device is powered off
App connect error callback is executed so App notices that device has gone away
App goes back to searching for device 

In this scenario the call to "scan" never finds the device again.  

Another possibility, of course, is that the BT stack on the phone the App is running on still thinks there is a connection to this device.  Is that possible?  I do call "disconnect" on the device even in the situation where the connection is lost, just in case a clean up was needed.

Anyway, help?

don, 2015-10-15:
Usually when devices don't show in a scan, I find that they it still connected to something else and not advertising. If you power off a peripheral and you receive the failure callback from the connect method, you should not need to call disconnect.

What version of Android are you seeing this on? Can you duplicate this across multiple versions of Android or on different Android hardware? Does this happen with any peripheral or specific peripherals? It's possible that Android isn't returning the peripheral in the scan because it thinks it's a duplicate.

don, 2015-10-23:
@halindrome if you have more info, add it here and I can re-open the issue if necessary

closed 2015-10-23 by don