don/cordova-plugin-ble-central #109: Call to scan() function on iOS device causes crash

matdev, 2015-10-20:
Hello,

It occurs on device running iOS 9.0.1

Has anyone experienced the same problem ?

The Android version of the ionic app works fine.

don, 2015-10-21:
Can you supply some code and stack traces?

JackPrice, 2015-10-29:
I've seen this happen too - I'll get a stack trace as soon as I can.

It happens only when scanning finds one or more peripherals. When the search would return no results, it doesn't crash.
 It doesn't happen on my iPhone 4S running 9.0.1, but does on an iPhone 6 running 9.0.1.

don, 2015-11-13:
@matdev @JackPrice are you still seeing this? A stack trace would help me fix it.

Max2817, 2015-12-07:
Hi don,

I have the same problem on my side. It crashes on Xcode here: 
#pragma mark - CBCentralManagerDelegate

- (void)centralManager:(CBCentralManager *)central didDiscoverPeripheral:(CBPeripheral *)peripheral advertisementData:(NSDictionary *)advertisementData RSSI:(NSNumber *)RSSI {

    [peripherals addObject:peripheral];
    [peripheral setAdvertisementData:advertisementData RSSI:RSSI];

    if (discoverPeripherialCallbackId) {
        CDVPluginResult *pluginResult = nil;
        pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsDictionary:[peripheral asDictionary]];
        NSLog(@"Discovered %@", [peripheral asDictionary]);
        [pluginResult setKeepCallbackAsBool:TRUE];
        [self.commandDelegate sendPluginResult:pluginResult callbackId:discoverPeripherialCallbackId];
    }

}

At this line 
[self.commandDelegate sendPluginResult:pluginResult callbackId:discoverPeripherialCallbackId];

discoverPeripherialCallbackId is nil then it crashes during the Json serialization.

matdev, 2015-12-16:
Hi Don,

Do you think the bug could be fixed anytime soon ?

This article mentions crashes occurring on iOS 9 with BLE, even though I noticed your plug-in no longer uses the deprecated methods:
http://www.argenox.com/blog/whoops-ios-9-crashes-bluetooth-low-energy-apps/

don, 2015-12-16:
@matdev Can you send a stack trace from iOS? What's in the peripheral name and id? Do you know what peripheral is causing this?

I can't reproduce this, so perhaps particular peripherals are creating data that is crashing the app? I'll need to figure out how to code around it.

@Max2817 Can you verify `discoverPeripherialCallbackId` is nil? The code shouldn't send the plugin result if didDiscoverPeripheral is nil.

matdev, 2016-01-13:
Hi @don , since we've upgraded XCode to version 7.2, we no longer have the problem, and our app is able to detect and communicate with our peripheral (a RedBearLab's BLE Nano).

Thanks for your help and your great plugin

don, 2016-03-19:
Assuming #168 fixed this

closed 2016-03-19 by don

chris-guidry, 2019-09-17:
Hi Don,

This error is occurring intermittently with our app/device. Here is the crash log: [BmsvR4r2LvJ8AkwHHcyey9.xccrashpoint.zip](https://github.com/don/cordova-plugin-ble-central/files/3621427/BmsvR4r2LvJ8AkwHHcyey9.xccrashpoint.zip)

Chris

chris-guidry, 2019-09-23:
This issue didn't used to occur at all, and now it's growing in frequency even though our BLE hardware/software has not been modified so I'm wondering if there are changes in iOS that are causing it and need to be handled differently? Any thoughts regarding error handling in that function?