don/cordova-plugin-ble-central #110: Writing returns an error which is just the number 13

RangerMauve, 2015-10-21:
Hi, I'm trying to talk to [one of these buttons](http://www.vsnmobil.com/products/v-bttn-wearable-bluetooth-le-4-0-device) and part of the setup requires that I write to a certain characteristic within 30 seconds of connecting to it. Specifically, it wants me to write to `fffffff5-00f7-4000-b000-000000000000` with `[0x80, 0xbe, 0xf5, 0xac, 0xff]` on the `fffffff0-00f7-4000-b000-000000000000` service.

My code flow looks something like this:
 * `ble.scan`
 * Chose a device from a list
 * `ble.connect()` with the id of the device
 * Wait for connect to be successful
 * `ble.write(device, "fffffff0-00f7-4000-b000-000000000000", "fffffff5-00f7-4000-b000-000000000000", new Uint8Array([0x80, 0xbe, 0xf5, 0xac, 0xff]), etc, etc)`
 * After a second or two the error callback gets called with the number `13`

I have no idea what 13 means in this situation and where to even look for what it could mean.
Any help would be very much appreciated.

RangerMauve, 2015-10-21:
Tracked my issue to [this](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L250) line in Android. This probably has nothing to do with the library so I'll go pester the button people directly. If someone _does_ have any insight, it'd still be very aprreciated.

don, 2015-10-21:
Error 13 is GATT_INVALID_ATTRIBUTE_LENGTH
https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#GATT_INVALID_ATTRIBUTE_LENGTH

The value you pass in should be an `ArrayBuffer` rather than `Uint8Array`. Use `.buffer` to get an ArrayBuffer from a typed array.

    var data = new Uint8Array([0x80, 0xbe, 0xf5, 0xac, 0xff]).buffer

don, 2015-10-21:
This is related to #64

RangerMauve, 2015-10-21:
Would it maybe make sense to have code that makes sense of these errors when possible and converts them to an Error object with a proper message?

RangerMauve, 2015-10-21:
Also thank you very much for pointing it out. I ended up figuring it out after a bit of struggle but this helps a lot as well!

closed 2015-10-22 by RangerMauve