don/cordova-plugin-ble-central #111: Writing a descriptor of a characteristic

schoebsam, 2015-10-22:
I am not sure if it is already supported somehow, but I require to write to a descriptor of characteristic.
If indeed it is possible to do so, it'ld be great to highlight it in the otherwise great API documentation.
Thanks for considering.

luposlip, 2015-10-22:
Your timing is exceptional.
Just today Bluetooth SIG released their BDS (Bluetooth Developer Studio)
1.0 for doing exactly that!

Check it out (unfortunately it's Windows only - for now at least):
http://www.bluetooth.com/Pages/Bluetooth-Developer-Studio.aspx

don, 2015-10-22:
@schoebsam currently there's no API for writing (or reading) descriptors.

When you subscribe for notification or indications on a characteristic, the library writes the the proper value to the client characteristic configuration descriptor (0x2902).

Both iOS and Android support writing to descriptors so this could be added to the API. I'm curious, what's your use case for writing descriptor values? This isn't something I've needed yet.

don, 2018-06-08:
When this is implemented, add read and write APIs for descriptors.