don/cordova-plugin-ble-central #112: Detect when scanned devices are out of range

RangerMauve, 2015-10-22:
Is there a way to detect that previously detected devices via a scan have gone out of range? Or should I just periodically stop the scan, clear the buffer, and attempt to find devices from scratch?

don, 2015-10-22:
For now you'll need to rescan.

With the latest Android you can get [CALLBACK_TYPE_MATCH_LOST](https://developer.android.com/reference/android/bluetooth/le/ScanSettings.html#CALLBACK_TYPE_MATCH_LOST) in [ScanCallback.onScanResult](https://developer.android.com/reference/android/bluetooth/le/ScanCallback.html#onScanResult(int, android.bluetooth.le.ScanResult)) so this is possible.

I'm not sure how to do this with older versions of Android or iOS. Maybe modify the plugin to adjust the scanning to return duplicate the results. Then you could remove items from your set if you haven't seen them in a while.

RangerMauve, 2015-10-22:
Yeah, might make sense to wait a year or two, then. :o

closed 2015-10-23 by don