don/cordova-plugin-ble-central #114: Possible conflict with the BluetoothSerial plugin?

jernejc, 2015-10-31:
Hi

Thanks for the plugin(s)!

iPhone 5s (iOS 8)
Currently testing on module BC127 from BlueCreation:
https://www.bluecreation.com/product_info.php?products_id=38

I start the BLE scan with an empty array and 10 second time interval, but it seems like after the module is detected I get the following error:

*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** setObjectForKey: object cannot be nil (key: advertising)'
*** First throw call stack:
(0x2da05ecb 0x381a0ce7 0x2d94117f 0xd713d 0xcf203 0x2d727bb3 0x2d728083 0x2d73066f 0x38689d53 0x38689d3f 0x3868c6c3 0x2d9d0641 0x2d9cef0d 0x2d939729 0x2d93950b 0x328a86d3 0x3029a871 0xcc13b 0x3869eab7)
libc++abi.dylib: terminating with uncaught exception of type NSException

The weird part is, that the error is coming for the BluetoothSerial plugin, which I'm using for Android SPP connections, but trying to fallback on this plugin for iOS BLE connections (the BluetoothSerial plugin didn't list the module, despite being paired/connected to the device).

The scan is definitely initialized by the BLE central plugin, but the error comes from:
cordova-plugin-bluetooth-serial/src/ios/CBPeripheral+Extensions.m (line 39)

Any ideas? I can't really remove the serial plugin as it's working great for Android and this seemed like a nice fallback for iOS.

jernejc, 2015-10-31:
After adding Serial and Characteristic UUIDs in the Serial plugin, it worked nicely. But this plugin still crashes on scan as mentioned in the edit above.

Thanks again for these, really nice!

don, 2015-10-31:
Both plugins extend CBPeripheral, so that's likely part of the problem. The last one loaded would win. I probably need to rename the functions so they can co-exist.

I wonder if the crash you're seeing is the same as #109? It's likely I'm not handling the advertising information correctly for some devices.

What version of the plugin are you using? I'm not seeing an `else` statement on [line 39](https://github.com/don/cordova-plugin-ble-central/blob/a16b1746cba3292e5eb2f2b026cfbd465ea59c5f/src/ios/CBPeripheral%2BExtensions.m#L39)

If you're still see the error on [line 307](https://github.com/don/cordova-plugin-ble-central/blob/a16b1746cba3292e5eb2f2b026cfbd465ea59c5f/src/ios/BLECentralPlugin.m#L307) it's likely the monkey patched version of CBPeripheral is not loading correctly. Try uninstalling both plugins with cordova cli then re-installing this plugin.

don, 2015-12-03:
Fixed the conflict in version 0.4.5 of [cordova-plugin-bluetooth-serial](https://www.npmjs.com/package/cordova-plugin-bluetooth-serial)

closed 2015-12-03 by don