don/cordova-plugin-ble-central #115: Android 5.x can't connect to radio

bardu, 2015-11-03:
I'm running cordova 5.1.1 and the latest version of the plugin. I'm having trouble to connect to a ble radio from a Samsung S6 Edge after a successful scan. Here is the output of the logcat:

> 11-02 17:27:05.035 20179-20511/ca.mobitec.sled D/BluetoothAdapter: startLeScan(): null
> 11-02 17:27:05.085 20179-20193/ca.mobitec.sled D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=8
> 11-02 17:27:05.085 20179-20511/ca.mobitec.sled W/PluginManager: THREAD WARNING: exec() call to BLE.scan blocked the main thread for 58ms. Plugin should use CordovaInterface.getThreadPool().
> 11-02 17:27:11.725 20179-20193/ca.mobitec.sled D/ScanRecord: parseFromBytes
> 11-02 17:27:11.725 20179-20193/ca.mobitec.sled D/BluetoothLeScanner: onScanResult() - ScanResult{mDevice=20:CD:39:9E:38:9E, mScanRecord=ScanRecord [mAdvertiseFlags=6, mServiceUuids=[0000ffd0-0000-1000-8000-00805f9b34fb], mManufacturerSpecificData={}, mServiceData={}, mTxPowerLevel=4, mDeviceName=null], mRssi=-72, mTimestampNanos=161781970165496}
> 11-02 17:27:11.775 20179-20179/ca.mobitec.sled I/chromium: [INFO:CONSOLE(60)] "{"id":"20:CD:39:9E:38:9E","advertising":{},"rssi":-72}", source: file:///android_asset/www/index.html (60)
> 11-02 17:27:11.775 20179-20179/ca.mobitec.sled I/chromium: [INFO:CONSOLE(61)] "ID: 20:CD:39:9E:38:9E", source: file:///android_asset/www/index.html (61)
> 11-02 17:27:15.095 20179-20511/ca.mobitec.sled D/BluetoothAdapter: stopLeScan()
> 11-02 17:27:22.755 20179-20179/ca.mobitec.sled D/ViewRootImpl: ViewPostImeInputStage ACTION_DOWN
> 11-02 17:27:22.955 20179-20179/ca.mobitec.sled I/chromium: [INFO:CONSOLE(88)] "btn id 20:CD:39:9E:38:9E", source: file:///android_asset/www/index.html (88)
> 11-02 17:27:22.955 20179-20511/ca.mobitec.sled D/BluetoothGatt: connect() - device: 20:CD:39:9E:38:9E, auto: false
> 11-02 17:27:22.955 20179-20511/ca.mobitec.sled D/BluetoothGatt: registerApp()
> 11-02 17:27:22.955 20179-20511/ca.mobitec.sled D/BluetoothGatt: registerApp() - UUID=f0a5d7bb-1ccb-4319-b7ee-30240dd2734d
> 11-02 17:27:23.005 20179-20511/ca.mobitec.sled W/PluginManager: THREAD WARNING: exec() call to BLE.connect blocked the main thread for 47ms. Plugin should use CordovaInterface.getThreadPool().
> 11-02 17:27:23.005 20179-20189/ca.mobitec.sled D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
> 11-02 17:27:28.015 20179-20193/ca.mobitec.sled D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=8 device=20:CD:39:9E:38:9E
> 11-02 17:27:28.015 20179-20193/ca.mobitec.sled D/BluetoothGatt: close()
> 11-02 17:27:28.025 20179-20193/ca.mobitec.sled D/BluetoothGatt: unregisterApp() - mClientIf=8
> 11-02 17:27:28.035 20179-20179/ca.mobitec.sled I/chromium: [INFO:CONSOLE(96)] "could not connect to ble:[object Object]", source: file:///android_asset/www/index.html (96)
 
Does that help to identify the possible problem?

bardu, 2015-11-03:
Randomly it looks like a connection is established but fails after a read:

> 11-02 18:11:24.695 11321-11605/ca.mobitec.sled D/BluetoothGatt: connect() - device: 20:CD:39:9E:38:9E, auto: false
> 11-02 18:11:24.695 11321-11605/ca.mobitec.sled D/BluetoothGatt: registerApp()
> 11-02 18:11:24.695 11321-11605/ca.mobitec.sled D/BluetoothGatt: registerApp() - UUID=2d2d4757-d65e-4116-8f0b-d0f9b6c2d855
> 11-02 18:11:24.735 11321-11605/ca.mobitec.sled W/PluginManager: THREAD WARNING: exec() call to BLE.connect blocked the main thread for 46ms. Plugin should use CordovaInterface.getThreadPool().
> 11-02 18:11:24.735 11321-11578/ca.mobitec.sled D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
> 11-02 18:11:25.505 11321-11578/ca.mobitec.sled D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=20:CD:39:9E:38:9E
> 11-02 18:11:25.505 11321-11578/ca.mobitec.sled D/BluetoothGatt: discoverServices() - device: 20:CD:39:9E:38:9E
> 11-02 18:11:28.545 11321-11340/ca.mobitec.sled D/BluetoothGatt: onSearchComplete() = Device=20:CD:39:9E:38:9E Status=0
> 11-02 18:11:28.575 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(92)] "peripheral object", source: file:///android_asset/www/index.html (92)
> 11-02 18:11:28.575 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(93)] "[object Object]", source: file:///android_asset/www/index.html (93)
> 11-02 18:11:28.585 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(107)] "read from device 20:CD:39:9E:38:9E", source: file:///android_asset/www/index.html (107)
> 11-02 18:11:28.695 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(110)] "got tension:", source: file:///android_asset/www/index.html (110)
> 11-02 18:11:28.705 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(112)] "last tension: 545", source: file:///android_asset/www/index.html (112)
> 11-02 18:11:29.595 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(107)] "read from device undefined", source: file:///android_asset/www/index.html (107)
> 11-02 18:11:29.595 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(117)] "Could not read from device Peripheral null not found.", source: file:///android_asset/www/index.html (117)
> 11-02 18:11:29.605 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(119)] "Disconnect after read failure", source: file:///android_asset/www/index.html (119)

Is this an indication for a malfunctioning radio?

don, 2015-11-03:
Maybe the device disconnected? It looked like thing were working, then the device is undefined.

    11-02 18:11:29.595 11321-11321/ca.mobitec.sled I/chromium: [INFO:CONSOLE(107)] "read from device undefined", source: file:///android_asset/www/index.html (107)

Can you debug on the peripheral side to see what's happening?

bardu, 2015-11-03:
Thanks Don. The company I'm working with will build a new device with a newer, different peripheral. For now I just can wait and see. I will close this thread for now.

closed 2015-11-03 by bardu