don/cordova-plugin-ble-central #116: "ble is not defined", cordova-plugin-ble-central 1.0.2 "BLE" with Ionic

andreribeiropinto104, 2015-11-03:
Hi, 
I am conducting some tests with its ionic -ble example and when I try to run for this error appears .

don, 2015-11-03:
"ble is not defined" typically means that the plugin is not installed correctly

Ensure you've installed the plugin with 

    cordova plugin add cordova-plugin-ble-central

If you're using PhoneGap Build, you need to install the plugin in config.xml

    <gap:plugin name="cordova-plugin-ble-central" source="npm" />

See https://github.com/don/ionic-ble for a sample ionic project that uses this plugin

johnyso, 2015-11-12:
I had the same problem. I was working on another "older" git branch (without the plugin) after i switched back the plugin was there but maybe some references were missing. 
After running 
```
ionic platform rm android
ionic platform add android
```
it worked again.
the following command should solve this too:
```
ionic state restore
```

redphoenix2k, 2015-12-01:
Hi guys, i have a similar issue but without ionic

if i call ble functions directly in javascript with the phonegap debugger it works, but if i call ble functions inside an angular controller or service i get the undefined error as well

don, 2015-12-01:
@redphoenix2k if you can call the `ble` object from JavaScript, this sounds like a problem with the Angular code. Make sure you're not calling any Cordova APIs until after the deviceready event fires.

redphoenix2k, 2015-12-01:
I quickly made a small test app to showcase my error

This is all the code & i still get the error in angular when testing with phonegap developper app on android

vanilla javascript calls it just fine

Any idea what the problem might be ?

[image]

redphoenix2k, 2015-12-01:
ok i updated the test app a bit

i added a button to scan directly with javascript and fixed a typo in the previous code (it was wrong, i now put the ble device id string in proper brackets [])

you can see better from this second screenshot how one call gives the error and another doesnt

[image]

redphoenix2k, 2015-12-02:
@don issue solved

the problem was that when using angular i must explicitly add before it  <script type="text/javascript" src="cordova.js"></script>

doing so made the ble plugin work the same as in pure javascript

[image]

closed 2015-12-03 by don

vinimendes3, 2016-03-04:
@andreribeiropinto104 Did you solve your problem? I'm with the same problem here...

sharathr3, 2016-07-24:
I m facing same issue here.. "ble not defined"  Guys, have you fixed ?