don/cordova-plugin-ble-central #118: "WriteWithoutResponse" Cause my page reloads

yangheng, 2015-11-09:
First of all ,the front framework is angularjs and Ionic. after I use "WriteWithoutResponse" to send data to device and device had received data but my page reloads, I don't know why.Everything is ok  when I use “startNotification” listen to device . Can you help me ?

don, 2015-11-09:
Are you seeing this on iOS or Android? Is the value being written to the characteristic on the peripheral side? Do you know if the success or failure callback for ble.writeWithoutResponse is begin called?

My guess is that there's a JavaScript error that is causing the page to reload.

closed 2015-12-03 by don

yangheng, 2015-12-21:
Yes,you are right ! It's JavaScript error ,thank you !

don, 2015-12-21:
@yangheng thanks for letting me know. Glad you have it working.