don/cordova-plugin-ble-central #119: UART like service

zabojad, 2015-11-09:
Hi @don !

I'm trying to interact with a UART prototype.

I've read the examples with UART like services you've made: rfduinoLedButton, bluefruitle and redbearlab.

I have problems trying to setup my client cordova app for this UART service but I'm not sure what I'm doing wrong so here are a couple of questions:

 - First, on iOS, the lib prints the following in XCode debug logs:
```
advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = "WB_UART_DEMO";
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "V0II6+0AAAM=";
        };
        kCBAdvDataServiceUUIDs =         (
            FFF0
        );
    };
    id = "A0A024DF-A384-A45C-05FD-1DA6D9646E57";
    name = "WB_UART_DEMO";
    rssi = "-75";
Found characteristics for service <CBService: 0x1759d3b0, isPrimary = YES, UUID = FFF0>
Characteristic <CBCharacteristic: 0x1764dae0, UUID = FFF4, properties = 0x12, value = (null), notifying = NO>
Characteristic <CBCharacteristic: 0x176a2f20, UUID = FFF5, properties = 0x8, value = (null), notifying = NO>
```
I have almost no doc for the device I'm trying to interact with. Does this mean there is only one service with 2 characteristics? Then, the definition of my UART service should probably be this below, is that correct?
```
// my undocumented :( UART service
var mysrv = {
    serviceUUID: "FFF0",
    txCharacteristic: "FFF4",
    rxCharacteristic: "FFF5"
};
```

 - When I try to write to one of those characteristics, using both write() or writeWithoutResponse(), It doesn't work and I get (in Android console logs) from the failure callback:
```
Characteristic 0000fff4-0000-1000-8000-00805f9b34fb not found.
```

Here is are the only explanations regarding the format of data I'm supposed to write/pass to the machine:
```
@Override
	public void onClick(View v) {
		String order = "";
		if (v == btn1) {
			//send hex data
			// System_index--0x02
			// Mode_index0--0x03
			// Mode_index1--0x00
			// SetTimeH--0x00
			// SetTimeL--0x3c
			// SetSpeed--0x05
			// SetTemp--0x64
			order = "020300003C0564";
		}
		final Intent intent = new Intent(action_data);
		intent.putExtra("order", order);
//		new Handler().postDelayed(new Runnable() {
//			
//			@Override
//			public void run() {
				sendBroadcast(intent);				
//			}
//		}, 10*1000);
	}
```

I've tried to pass this piece of data (020300003C0564) in different forms:
 - first I filled a Uint8Array witht he values [0x02,0x03,0x00,0x00,0x3C,0x05,0x64] and passed the ```.buffer``` property of that Uint8Array to the ble.write method

 - then I tried to pass ```stringToBytes("020300003C0564")``` with stringToBytes the same function as in your bluefruitle example...

And on both characteristics FFF4 and FFF5...

Each time, I endup on the write error callback with the same message...

Any hint for me to find a way to make it work?

Thanks,
Thomas

don, 2015-11-09:
@zabojad When you connect to the device, you should get a JSON object back that describes the peripheral in the success callback. Can you print that object out and paste the results here?

    JSON.stringify(peripheral, null, 2);

zabojad, 2015-11-09:
@don, this is what I get on iOS:
```
{"name":"WB_UART_DEMO","id":"A0A024DF-A384-A45C-05FD-1DA6D9646E57","advertising":{"kCBAdvDataLocalName":"WB_UART_DEMO","kCBAdvDataIsConnectable":true,"kCBAdvDataServiceUUIDs":["FFF0"],"kCBAdvDataManufacturerData":{}},"services":["FFF0"],"characteristics":[{"service":"FFF0","isNotifying":false,"characteristic":"FFF4","properties":["Read","Notify"]},{"service":"FFF0","isNotifying":false,"characteristic":"FFF5","properties":["Write"]}],"rssi":-83}
```

zabojad, 2015-11-09:
From this, I understand I should write on characteristic FFF5 but when I do so, I get this (here on iOS):
```
Could not find characteristic with UUID FFF5 on service with UUID FFF0 on peripheral with UUID A0A024DF-A384-A45C-05FD-1DA6D9646E57
```

zabojad, 2015-11-09:
I do not understand why, but the same thing on Android outputs much more information:
```
{
  "name": "WB_UART_DEMO",
  "id": "08:EB:ED:00:00:03",
  "advertising": {},
  "rssi": -78,
  "services": [
    "1800",
    "1801",
    "fff0"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a01",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a02",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a03",
      "properties": [
        "Write"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a04",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1801",
      "characteristic": "2a05",
      "properties": [
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "fff0",
      "characteristic": "fff4",
      "properties": [
        "Read",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        },
        {
          "uuid": "2901"
        }
      ]
    },
    {
      "service": "fff0",
      "characteristic": "fff5",
      "properties": [
        "Write"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        }
      ]
    }
  ]
}
```

zabojad, 2015-11-09:
hmmm... Writing an ArrayBuffer containing [0x02,0x03,0x00,0x00,0x3C,0x05,0x64] on 1800/2a02 fails with message ```00002a02-0000-1000-8000-00805f9b34fb not found.```

And writing the same thing to 1800/2a03 fails too with message: ```Characteristic 00002a03-0000-1000-8000-00805f9b34fb not found.```

What do I do wrong ?

zabojad, 2015-11-09:
@don I'm completely stuck here, so you see anything I do wrong? Or anything I could try?

don, 2015-11-09:
@zabojad based on the JSON you posted, you should be writing to FFF5. It should be finding the characteristic when you call ble.write(...). I'm not sure why it's not working. 

@kanayo had some problematic devices and submitted a pull request #113 that might help you. I'm going to try and merge his code later this week.

You're seeing more info on Android since iOS hides some of the standard GATT services during discovery.

zabojad, 2015-11-09:
@don It nows work when I write to FFF5 (at least the success cb of ble.write() is called). I've messed up my code a little bit when trying different things and I was actually using writeWithoutResponse() instead of write(). In my case, it seems I should use write().

What are the ```descriptors``` I see in my Android peripheral data output ?
```
{
      "service": "fff0",
      "characteristic": "fff4",
      "properties": [
        "Read",
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        },
        {
          "uuid": "2901"
        }
      ]
    },
```
How can we use them? (those two uuid "2902" and "2901")

don, 2015-11-09:
Im glad you got this to work. Once I get #113 integrated, you'll get a different error when you try and writeWithoutResponse to a characteristic that only supports write. The characteristic's properties control how you can access them.

You can't do much with descriptors from this plugin. Some day I'll add methods to read & write descriptors.

[0x2901](https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.characteristic_user_description.xml)  is the user description. You can see this in tools like LightBlue for iOS. It displays a friendly name instead of the UUID 0xFFF5. 

[0x2902](https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml) is the client characteristic configuration. If you call ble.startNotification, the plugin writes values to this descriptor to subscribe for notifications.

zabojad, 2015-11-09:
Thanks a lot @don for your answers and explanations... And for this cordova plugin of course ^^!

don, 2015-11-10:
@zabojad you're welcome

closed 2015-11-10 by don

zabojad, 2015-11-10:
@don one last thing: on my prototype (the machine itself), nothing happens when I write on FFF5 (even though the success callback of ble.write() is called and no error show up). Would the error callback of ble.write() called if I pass data to ble.write() not in the format expected by the machine ?

zabojad, 2015-11-12:
Hi @don ! I've sent you an email today, can you please confirm you've received it and that it did not endup in your spam folder ?
Thanks in advance :) !
Best,
Thomas