don/cordova-plugin-ble-central #120: Crash on scan on iOS

zabojad, 2015-11-09:
```
2015-11-09 15:49:15.511 MyProto[945:327151] Cordova BLE Central Plugin
2015-11-09 15:49:15.512 MyProto[945:327151] (c)2014-2015 Don Coleman
2015-11-09 15:49:15.514 MyProto[945:327151] [CDVTimer][ble] 3.005028ms
2015-11-09 15:49:15.516 MyProto[945:327151] Bluetooth Serial Cordova Plugin - BLE version
2015-11-09 15:49:15.516 MyProto[945:327151] (c)2013-2014 Don Coleman
2015-11-09 15:49:15.520 MyProto[945:327151] [CDVTimer][bluetoothserial] 4.830956ms
2015-11-09 15:49:15.520 MyProto[945:327151] [CDVTimer][TotalPluginStartup] 9.041011ms
2015-11-09 15:49:15.660 MyProto[945:327151] Status of CoreBluetooth central manager changed 5 State powered up and ready (CBCentralManagerStatePoweredOn)
2015-11-09 15:49:15.660 MyProto[945:327151] Status of CoreBluetooth central manager changed 5 (State powered up and ready (CBCentralManagerStatePoweredOn))
2015-11-09 15:49:15.891 MyProto[945:327151] Resetting plugins due to page load.
2015-11-09 15:49:16.093 MyProto[945:327151] Finished load of: file:///private/var/mobile/Containers/Bundle/Application/73027F1D-B17E-4704-A5BE-B107AA2A8A13/MyProto.app/www/index.html
2015-11-09 15:49:22.942 MyProto[945:327151] scan
2015-11-09 15:49:23.028 MyProto[945:327151] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** setObjectForKey: object cannot be nil (key: advertising)'
*** First throw call stack:
(0x2959449f 0x36d4ac8b 0x294b1c63 0xee251 0xe5bc9 0x292680b7 0x292685b7 0x29273357 0x25c9db 0x25c9c7 0x264e29 0x25f2c9 0x26021f 0x2955a3b1 0x29558ab1 0x294a63c1 0x294a61d3 0x308a40a9 0x2cab5fa1 0xe270b 0x372caaaf)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) 
```

Any idea why ?

zabojad, 2015-11-09:
mmm it might be because I still had the bluetooth serial plugin installed too... I"ve cleant everything and it doesn't crash anymore...

closed 2015-11-09 by zabojad

don, 2015-11-09:
Unfortunately this plugin doesn't work well with BluetoothSerial. See #114 

I'm going to look into fixing this for future releases.

xam8re, 2016-04-24:
Same crash here. But I haven't Bluetooth serial installed.

don, 2016-04-24:
@xam8re the problem for this issue was solved fixed, it was a compatibility issue with Bluetooth Serial. 

If you're seeing problems, open a new issue. Include more details: OS, phone, Cordova, and plugin versions. Include how to reproduce & output from Xcode and/or the browser console.