don/cordova-plugin-ble-central #124: Problem seaching for a device on iOS

hhlevnjak, 2015-11-11:
Hi, 

 I'm having a problem searching for a device using this plugin, it is a BLE 4.0, the chip is DX-BT05.

I can find it and connect to it on an android phone using this plugin, but searching for it doesn't give anything back on iOS 8.4. Perhaps there is something I can enable or change in the configuration in order for it to work on iOS?
I also found that I can get all the service characteristics using the LightBlue application on iOS, so it should be possible to connect to it from the same device somehow...

I also tried the https://github.com/don/BluetoothSerial plugin, it works great for a bluetooth 2.0 device on android, but works on neither device with the BLE chip mentioned above.I tried editing the BLEDefines file there, with no result (or perhaps I didn't do it correctly).

Perhaps you are familiar with a similar problem, or I can make the serial plugin work somehow?

Thanks in advance,
Hrvoje

hhlevnjak, 2015-11-11:
It seems that I managed to resolve the issue, but I'm not sure how or what I did to do it.

I've seen that there are some other reported issues about compatibility between bluetoothSerial and this plugin. I did have both of them included at some point, so that might be the same issue as it is there, but I didn't get any error from it...

closed 2015-11-11 by hhlevnjak

don, 2015-11-13:
@hhlevnjak I expect the problem was having both plugins installed. I hope to fix that conflict in future releases.

hhlevnjak, 2015-11-19:
Thank you, I guess that must be it.

I have another issue, not sure if it is plugin related or is it an iOS issue, or perhaps the bluetooth device itself, so not to open a new one, I'll ask here.

On iOS, when I connect to a device from the application, it asks for the PIN to be entered, and that is ok, but is there a way to force the device to stay remembered in the device settings, so that you don't need to enter the PIN every time you need to reconnect?

Like I said, I don't know if this is even possible and whether it is a plugin issue.

don, 2015-11-19:
@hhlevnjak new issues are preferred for different topics. iOS is asking for the pin since some characteristics have require encrypted read/write. You *might* be able to bond the device by pairing in the iOS Bluetooth screen.

hhlevnjak, 2015-11-20:
I can open a new issue, I just wasn't sure if this is an issue with the plugin...

Pairing from the settings does not work for some reason.