don/cordova-plugin-ble-central #126: Android Scan > Connect > BackgroundApp > Scan doesn't include peripheral in results

JohnMcLear, 2015-11-11:
If I do a 
1. ``.scan``
1. ``.connect`` to a device
1. Exit the app (Android puts the app in the background) 
1. Go back into the app
The Device I was connected to doesn't show up in a ``.scan`` and doing 

```
ble.isConnected(
    id,
    function() {
        console.log("Peripheral is connected");
    },
    function() {
        console.log("Peripheral is *not* connected");
    }
);
```
Returns
```
Peripheral is *not* connected 
```
``.disconnect`` also makes no difference.

If I kill the app then go back into it, everything works fine...

don, 2015-11-18:
I can replicate this, but I'm not sure how to fix it. For now I think a back button event listener is the best work around.

    document.addEventListener("backbutton", onBackButton, false);

    function onBackButton() {
        // if connected, disconnect before exiting app
    }

JohnMcLear, 2015-11-19:
I assume ``onBackButton`` fires on hardware back button too?

don, 2015-12-02:
@JohnMcLear yes, it fire with the hardware back button

don, 2015-12-02:
Maybe future versions of the plugin can provide an API to find connected devices with [CBCentralManager retrieveConnectedPeripheralsWithServices](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html#//apple_ref/occ/instm/CBCentralManager/retrieveConnectedPeripheralsWithServices:) and   [BluetoothManager.getConnectedDevices](http://developer.android.com/reference/android/bluetooth/BluetoothManager.html#getConnectedDevices(int)).

thomasmcneill, 2017-01-23:
Is there any update on this?  My application stores the last UUID so could issuing a disconnect when the application opens resolve this?  I'd love to help on the back end but the reason I use phonegap build is so I don't have to have OSX to build.

don, 2017-02-02:
@thomasmcneill I looked into this a while back. I didn't see a great cross platform way to implement it. I make sure I have hooks to disconnect the device on the back button so it'll show up in future scans. I'm basically ignoring the issue for now until I need it in a project or someone hires me to implement for them.

don, 2018-06-11:
I think this is the app's responsibility to keep track of the device or disconnect. On Android, you can connect using a MAC address without scanning. This should allow you to re-connect to lost devices. This probably works fine if you hold onto the UUID on iOS, as long as the plugin still has a reference to the CBPeripheral. You can also try [ble.peripheralsWithIdentifiers](https://github.com/don/cordova-plugin-ble-central#peripheralswithidentifiers) or [ble. connectedPeripheralsWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices) on iOS.

closed 2018-06-11 by don