don/cordova-plugin-ble-central #127: Howto connect to BTLE device using PIN

hohae, 2015-11-23:
I want to connect to a BTLE device which requires a PIN. How can I achieve that?

don, 2015-11-25:
If the device required a passkey for some characteristics, iOS or Android, should prompt you when necessary.

You *might* be able to bond to device using the Bluetooth settings of iOS or Android to avoid the prompt while running the app.

hohae, 2015-11-26:
Do you have experience with setting the PIN from the app? In my use case I'll be using static PINs or later on PINs coming from a database.
I've seen that at least on Android there are some methods like SetPin().

don, 2015-11-30:
There's no support in the plugin for setting a pin. If you want this functionality, you'll need to look into the native APIs.

closed 2015-12-02 by don

gwayne, 2015-12-28:
I never get a secure PIN prompt in my Cordova iOS app (mostly based on the senortag demo) when connecting to my own custom embedded peripheral with encrypted attributes.  When connecting to the same peripheral using the LightBlue test application, prompt to enter the PIN is presented on an initial connection attempt, as it should.  Thereafter bonding info is automatically recalled on connection and no PIN is needed.  It's the same for a purely native version of the app created in ObjC.  The Cordova app completes the initial GAP connection, but never sends a secure parameters request to the peripheral to initiate a secure connection.

don, 2015-12-29:
What happens if you try and read or write a characteristic that requires encryption? Are you seeing an error? In the past, I'm pretty sure, I get prompted for a PIN for elevated encryption. (I don't have a device to test right now.)

If you your app or LightBlue bonded with the peripheral, I would not expect to get prompted when reading the characteristic.

I haven't done a lot with encrypted characteristics, so there are probably issues. If you could help get repeatable test case that would be helpful for improving this plugin.

gwayne, 2015-12-29:
Adding a "read" operation  did it.  I had the characteristics set simply to notify on connect without ever calling a read/write function.  Adding a call to "read" just one of the characteristics before enabling notifications, triggered the PIN prompt.

FYI, somewhat related, I initially struggled getting the prompt to work in a native app as well.  Eventually I found that in addition to removing a previous connection to the device (made prior to adding encryption) in the iOS BT settings menu ("forget this device"), bluetooth needed to be turned off, and then on again, before it detected encryption and presented the PIN prompt.

Thanks for your work and taking time to respond.