don/cordova-plugin-ble-central #129: Android BLE connection crashes after 15-25 seconds

pekeboy, 2015-11-24:
BLE connection crashes after 15-25 seconds. Connection and data transferring works 100% until the crash.

I get errors:

ERROR: [object Object]
bleConnection.js (168,17)

Error in Error callbackId: BLE434748434 : TypeError: Cannot convert undefined or null to object
cordova.js (310,47)

Uncaught TypeError: Cannot convert undefined or null to object
cordova.js (312,0)

I tested with Motorola LTE G, 5.1 lollipop and Nexus 7 6.0 Marshmellow. I am using Visual Studio Cordova software. Device I am connecting is our own device with BLE chip. It is working with iPhone and also other devices so I don´t think there is a problem with a chip.

Any ideas?

don, 2015-11-25:
"Cannot convert undefined or null to object" sounds like it could be something in the data. It's odd that it works on some hardware and not others.

What function are you calling at bleConnection.js:168?

Can you run in a debugger and see what happens in the plugin before this crash? I need more info to figure out what's going on.

don, 2015-12-02:
@pekeboy ping

pekeboy, 2015-12-08:
Sorry for the delay. I´ve been superbusy in other projects.

I have tried this without sending or receiving any data. It doesn´t help. If I just open connection and leave it there, app crashes.

bleConnection.js:168 calls normal Error function from your library (onError).

cordova.js (310,47) and cordova.js (312,0) Catches this error.

callbackFromNative: function(callbackId, isSuccess, status, args, keepCallback) {
        try {
            var callback = cordova.callbacks[callbackId];
            if (callback) {
                if (isSuccess && status == cordova.callbackStatus.OK) {
                    callback.success && callback.success.apply(null, args);
                } else if (!isSuccess) {
                    callback.fail && callback.fail.apply(null, args);
                }
                /*
                else
                    Note, this case is intentionally not caught.
                    this can happen if isSuccess is true, but callbackStatus is NO_RESULT
                    which is used to remove a callback from the list without calling the callbacks
                    typically keepCallback is false in this case
                */
                // Clear callback if not expecting any more results
                if (!keepCallback) {
                    delete cordova.callbacks[callbackId];
                }
            }
        }
        catch (err) {
            var msg = "Error in " + (isSuccess ? "Success" : "Error") + " callbackId: " + callbackId + " : " + err;
            console && console.log && console.log(msg);
            cordova.fireWindowEvent("cordovacallbackerror", { 'message': msg });
            throw err; // line 312
        }

don, 2015-12-08:
Can you run `adb logcat` and paste the Java stack track from the crash? I think something is happening before it reaches cordova.

"bleConnection.js:168 calls normal Error function from your library (onError)." Which API call is the error handler begin called for? ble.connect?

pekeboy, 2016-01-04:
Here is the whole debug file from starting the app and crash: https://drive.google.com/file/d/0B9QjSznhfSCfVjU1SjQzc2RmQVU/view?usp=sharing

don, 2016-04-15:
@pekeboy are you still seeing this issue? 

I'm not seeing much here. Maybe there's some Javascript that's failing on Android 5+ causing the null peripheral?  

    I/chromium(19947): [INFO:CONSOLE(168)] "ERROR: Peripheral null not found.", source: file:///android_asset/www/scripts/bleConnection.js (168)
    I/chromium(19947): [INFO:CONSOLE(310)] "Error in Error callbackId: BLE963726003 : TypeError: Cannot convert undefined or null to object", source: file:///android_asset/www/cordova.js (310)
    I/chromium(19947): [INFO:CONSOLE(3)] "Uncaught (in promise) TypeError: Cannot convert undefined or null to object", source: file:///android_asset/www/scripts/jquery-2.1.4.min.js (3)

If this is still a problem, can you email me a debug version of your apk to try and duplicate this issue on my devices?

pekeboy, 2016-04-18:
I have tested the connection with other BLE apps (native) also with android 5.0+ and they also crashes with the chip. The BLE chip is Texas Instruments CC2541. Your code works with other chips and devices.

don, 2016-04-18:
@pekeboy Thanks for the update. It sounds like this is a phone specific problem. 

Have you tried Radius Labs Bluetooth Crash Resolver?
    http://developer.radiusnetworks.com/2014/04/02/a-solution-for-android-bluetooth-crashes.html
    https://play.google.com/store/apps/details?id=com.radiusnetworks.bluetoothcrashresolver

closed 2016-08-19 by don