don/cordova-plugin-ble-central #131: ble.isConnected not working for iOS?

gwayne, 2015-12-01:
I am attempting to use ble.isConnected in an iOS project per the method definition and example but it never works.  Using Cordova CLI, latest version of your plugin, and Xcode 6.4.  The way I understand it when the method is called, if a peripheral is connected the "success" callback is called, otherwise the "failure" callback is invoked.  However, it always generates a "disconnected" or often nothing at all.  I placed NSLog entries within BLECentralPlugin.m->isConnected() for both the if/else connected/disconnected conditions.  But there's never any log output for either of them after calling ble.isConnected().  It's as if BLECentralPlugin.m->isConnected() is never invoked?  I see output from other NSlog entries (eg. stopScanTimer) just fine.  Perhaps I'm misunderstanding how this method works?

gwayne, 2015-12-02:
Update - BLECentralPlugin.m->isConnected does get called, the NSLog entries I added weren't in the platform working directory.  However the basic problem remains.

I'm trying to determine if a peripheral is already connected on startup before attempting to scan/connect.  When I connect to a peripheral, then restart the page/script the peripheral remains connected and Characteristics continue to update, as evidenced in the log console.  So there IS a peripheral connected, but the method call doesn't reflect/detect it.

gwayne, 2015-12-02:
Update - The problem is that the method doesn't use the service UUID to recognize a peripheral. iOS assigns it's own random UUID to a connected peripheral.  The method only works when passing it the iOS assigned UUID.  Thus, the method won't work for attempting to generically detect if a particular service UUID is connected.

don, 2015-12-02:
[ble.isConnected](https://github.com/don/cordova-plugin-ble-central#isconnected) reports if a **device** is connected, not a service.

iOS assigns UUIDs to devices that are scanned. You need to pass the UUID for a device to isConnected. The function works as intended, it just doesn't do what you want.

NOTE: you need to disconnect from the device before reloading the page otherwise you lose the reference and can't connect to the peripheral again unless you restart the app.

closed 2015-12-03 by don

gwayne, 2015-12-04:
Thanks for the clarification.  My app is comprised of several html pages for performing other tasks (eg access databases).  I had hoped there would be a way to maintain or recover a peripheral connection between page/screen transitions within the app, without disconnecting.  All the cordova BTLE examples I've seen thus far utilize a single html page and thus never have this problem.

Thanks again for taking time to respond.

don, 2015-12-04:
@gwayne Right now, I don't have a way to work across multiple page apps. (I write everything as single page apps, even with multiple tasks.)

It's *possible* an API could be added to get an existing connection (See https://github.com/don/cordova-plugin-ble-central/issues/126#issuecomment-161429513) but I haven't tried it.