don/cordova-plugin-ble-central #139: Not able to scan the Bluetooth devices

dinguluer, 2015-12-10:
Hello Don,

I have a bluetooth serial port device (funduino bluetooth Xbee) connected to my AVR controller
I am developing  android app using intel xdk & your plugin "cordova-plugin-ble-central"
I am testing app on my mobile phone HTC 516 Desire.

I am continously scanning for the device, just to test app is able to detect all the nearby bluetooth devices.
ble.startScan([], onScan, scanFailure);

But onscan()  function never calls. As i am not getting any alert from alert(JSON.stringify(peripheral));.

Please suggest what is wrong

Here is my javascript code :--
=================== JS code ================================
var app = {
    initialize: function() {
        this.bindEvents();
    },
    bindEvents: function() {
        document.addEventListener('deviceready', this.onDeviceReady, false);
    },
    onDeviceReady: function() {
        app.scan();
    },
    scan: function() {
        app.status("Scanning for Heart Rate Monitor");

        var foundHeartRateMonitor = false;

        function onScan(peripheral) {
            alert(JSON.stringify(peripheral));

        }

        function scanFailure(reason) {
            alert("BLE Scan Failed");
        }
              ble.startScan([], onScan, scanFailure);

    },

};

app.initialize();
=============================================

Regards,
Dinesh

don, 2015-12-10:
Dinesh, it looks like you have a JavaScript error. 

Output from `adb logcat`

    12-10 12:34:59.339 14618 14618 I chromium: [INFO:CONSOLE(12)] "Uncaught TypeError: app.status is not a function", source: file:///android_asset/www/js/index.js (12)

You can also see the error with chrome://inspect

[image]

If I comment out that line, your code works OK. (Tested with Cordova 5.4.1 on a Nexus 6)

dinguluer, 2015-12-11:
Hello Don,

Thanks for your quick reply.

I removed this bug & again compiled the project. But still i am not able to detect any near by device.

On device ready i use to get this alert, but no bluetooth device detected :--
```alert('Device is ready !!');```

I have tested with following mobile :---
HTC desire 516 - Android 4.3

I am having my intel_xdk project (Bluetooth_don_plugin.zip) & android app generated (bluetooth-comm.cordova.android.201512111057.apk) here :--

https://github.com/dinguluer/UiMagician/tree/master/testing_folder/Don_bluetooh

I have a sample android app from distributor from whome i buyed the bluetooth module.
From that app i am able to scan nearby bluetooth devices & able to connect to my bluetooth device. This means my hardware bluetooth device is working fine.

Can you see if this app works for your Nexus-6 mobile ?

Not getting what i am doing wrong.

Regards,
Dinesh

dinguluer, 2015-12-11:
Just one thing to point, at the end of this link for HTC bluetooth device:--
https://github.com/bcsphere/bluetooth
note: HTC use shared library to operate BLE device, you should add the 'com.htc.android.bluetooth.le.jar' in the build path manually to support HTC devices, 

I found the file here :--
https://github.com/bcsphere/bcfindme/tree/master/platforms/android/com.htc.android.bluetooth.le

Any idea do we need shared library for HTC device bluetooth ?

Regards,
Dinesh

don, 2015-12-11:
This worked OK on the Nexus 6.

Android 4.3 was the first version to get BLE. Things don't always work correctly there. It might be worse on your device if you need the proprietary HTC jar file. You could see if building with that jar file helps.

[image]

dinguluer, 2015-12-12:
Hello Don,

Thanks for your reply.

Just one more thing i will like to know.

>> You could see if building with that jar file helps.

Now if i have to build my intel-xdk cordova project with this JAR file then what i have to do.
Do i have to add this file in your plugin src folder ?
Means how to include this JAR file in my project so that it got build ?

1> Your plugin can communicate with BLE devices, means devices with Bluetooth version-4 right ?
   Can it communicate with devices with lower version < 4 ?

2> I am using this bluetooth device, HC-06 :--
http://www.order2india.com/products/58982_hc-06-arduino-bluetooh-bee-bluetooth-wireless-module?gclid=CjwKEAiAhaqzBRDNltaS0pW5mWgSJADd7cYDekZmlFMJhCDjY2qYBPrVhDnMgh2nSAx_88T5wxYRDRoCzTnw_wcB&variant=3724904324
   
   http://wiki.mikrokopter.de/en/HC-06

   As per specs it uses Bluetooth V2.0. So can your plugin communicate with this device ?

3> At my home i do not have embedded setup. I tried app at my home with my brothers mobile Google Moto-G having android version 5.0 (lolly pop) & it was not able to detect the near by sony srs-x2 speakers (Bluetooth ver. 3.0).

   So i have one doubt can your plugin communicate with devices with lower version < 4 ?  

   https://github.com/dinguluer/UiMagician/tree/master/testing_folder/Don_bluetooh/(bluetooth-comm.cordova.android.201512111057.apk
   http://www.sony.co.in/product/srs-x2

Please suggest.

Regards,
Dinesh

don, 2015-12-12:
That's the problem. The HC-06 is Bluetooth Classic. Try using cordova-plugin-bluetooth-serial https://github.com/don/BluetoothSerial instead of this plugin.

dinguluer, 2015-12-12:
Hello Don,

Thanks for your reply.

Please confirm this last point from my side.

Current plugin Cordova-plugin-ble-central which I am using will work only with Bluetooth version-4 right ?

Cordova-plugin-bluetooth-serial can work with Bluetooth version 2.0 & 3.0 right ?

I will test with this new plugin & update you.

Regards,
Dinesh

don, 2015-12-12:
Yes, this plugin is Bluetooth 4.0 only. The BluetoothSerial plugin uses Bluetooth 2.0 for Android & Bluetooth 4.0 for iOS.

dinguluer, 2015-12-15:
Hi Don,

Thanks for your support. 

I have just tested.
HC-05 bluetooth device is working fine with following plugin cordova-plugin-bluetooth-serial on my HCT 516 desire device.

I am able to send & receive messages from/to app & bluetooth module HC-05.

Regards,
Dinesh

closed 2015-12-15 by dinguluer

sumit1317, 2017-12-04:
I have Bluetooth 3.0 LE device which I want to search and connect via cordova app. Any pointers will be highly appreciable.
It is working perfectly fine with Android.

marianoabi, 2019-03-25:
Hi! Is this thread up-to-date?

I'm working on an ios project now that currently uses ble-central-plugin and a bluetooth printer that supports 2.0 and 4.0. Now, the client wants to change to a printer that supports bluetooth 2.0 only. But the app doesn't detect that new printer.

I saw in a thread that bluetoothSerial supports classic bluetooth while ble-central-plugin supports 4.0 and higher. But this is an old thread way back 2015 so I'm not sure if this is the issue that we've been encountering right now.

If anyone knows, please help. Thanks!

droghio, 2019-03-25:
@marianoabi Usually you should start a new thread instead of tacking onto a closed one, but not a big deal. This library only supports BLE which is a subset of Bluetooth 4+. If you are using Bluetooth 2 or Bluetooth classic this library will not work.