don/cordova-plugin-ble-central #140: Strange behavior with a Samsung Galaxy S3

H0rst, 2015-12-11:
Hello,

i have used the heartrate tracker example as base for a blood pressure app. It is working on different Android devices. But with a Samsung Galaxy S3 i have a strange behavior. I use also the nRF Master Control app for testing.

When i start my app and start scanning, no device is found. Now i start the nRF Master Control App. This app is scanning for devices at start. It also don't find a device. Now i click on "Stop scanning" in the nRF app. At this point, my app in the background connects to the device and receives the data.

When i stop the scan in my app, and go back to the nRF App, it is finding the device. The nRF app is working fine, when my app don't use Bluetooth or is not running.
But without the nRF App i never get a connection to the device. 

The issue i have tested with two blood pressure devices. And i have tested the app on 3 other Android devices and there are no problems.

Any idea?

best regards
Horst

don, 2015-12-11:
Is your app filtering scan results by UUID? I've had a lot of problems with filtering on Android-4.3 and Android-4.4. Can you test scanning for all devices, `ble.startScan([], success, failure)`, and see if it works better on the Galaxy S3?

H0rst, 2015-12-12:
Thank you. Scanning without service does the trick. Now i check for the services after connection. This works fine for me.
BTW: On a Huawei with Andriod 4.3 I did not have the problem.

closed 2015-12-12 by H0rst