don/cordova-plugin-ble-central #141: i cant find out device

ted66, 2015-12-14:
I follow this example
https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle

i want to connect Arduino BLE 4.0, but i can find out the device.
i have set UUID and run this comment 
ble.scan([bluefruit.serviceUUID], 5, app.onDiscoverDevice, app.onError);

it also not find any.
How could i check and connect it?

don, 2015-12-14:
Try scanning without any UUID `ble.scan([], 5, app.onDiscoverDevice, app.onError);`. Some Android devices have trouble filtering scan results by UUID.

ted66, 2015-12-15:
@don 
ok i have try it, but is alway
thx your reply

don, 2015-12-20:
@ted66 did the scan find your device without the UUID?

Make sure nothing is connected to you Bluetooth peripheral when scanning. Most device won't advertise when a central is connected.

closed 2016-03-19 by don