don/cordova-plugin-ble-central #142: Disconnect doesn't properly close connection on Android

tomasbedrich, 2015-12-17:
[From this method](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L74-L81) you can see only the `.close()` method is called. [The Android API provides](http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#disconnect()) a `.disconnect()` method. I think it should be called before `.close()`.

Currently I am experiencing strange issues (errors no. 133) when connecting and disconnecting a device few times in a row. [This SO post](http://stackoverflow.com/a/25364567/570503) led me to walk through your code and I discovered this potential problem. What do you think? I hope it can fix my issues.

don, 2015-12-17:
Older versions of this plugin used gatt.disconnect without gatt.close and that caused other issues. I though gatt.close called gatt.disconnect but maybe not on all versions of Android.

Can you modify your local copy to call gatt.close() before gatt.disconnect()? If it works better, submit a pull request.

If it doesn't work, get me some more details about your peripheral and your app and I'll see if I can duplicate it here.

tomasbedrich, 2015-12-20:
Unfortunately adding `gatt.disconnect()` hasn't solved my problem. I will try to further debug my code and post the conclusions here ASAP.

tomasbedrich, 2016-01-21:
It was a peripheral problem. Sorry for spam.

closed 2016-01-21 by tomasbedrich