don/cordova-plugin-ble-central #143: Windows support

lj222kj, 2015-12-18:
Hi! Is the windows-platform(phones) supported? :)

don, 2015-12-20:
Windows support is on the roadmap, but it's not high priority. Unfortunately Windows (and Windows Phone) idea of scanning and connecting is completely different from iOS and Android. I need to figure out how to reconcile the differences.

lj222kj, 2015-12-21:
Hey!
Thanks for getting back to me Don! 

Are you aware of any other library that supports all 3 major platforms? (Android, iOS, Windows phone) 

Thanks!

don, 2015-12-21:
I've heard they have Windows support. Have you tried it? Does it work well?

PeacePan, 2015-12-27:
hi,

I'm tried randdusing's plugin, but the api design is too complicated for me.

On Windows UWP platform, randdusing's plugin not implement scan and connect(pair) api.

I'm need these feature, because I don't want my users to operate bluetooth settings.

I'm forked this plugin to develop my Windows UWP, maybe you can refer my repository.

It used javascript to write. (depend on WinJS)

https://github.com/PeacePan/cordova-plugin-ble-central/tree/windows-dev/src/windows

It's still in developing

dmartinpro, 2016-04-08:
Any progress? Maybe W10 is more friendly?

VinceFa, 2017-03-02:
Hi ! I would like to know if Windows support is still planned, especially for Windows 10.

don, 2017-08-22:
See #368

closed 2017-08-22 by don