don/cordova-plugin-ble-central #144: Add ble.enable to iOS

papasmurph, 2015-12-19:
If it's not possible to inform the user to get Bluetooth activated, could you in that case instead always call the success function, so the same code works for both Android and iOS without errors?

Now, when Bluetooth is activated and I call ble.enable it still fails. I haven't checked whether the failure callback is called or there's a Javascript error (undefined) yet.

papasmurph, 2015-12-20:
After hooking an iPhone up to a Mac to see what's actually going on when ble.enable is being used:

There's no Javascript error, so the enable method is actually there (which it is; see below).

The success callback was never called.

I checked the plugin to see why:
- ble.js contains enable (and showBluetoothSettings) without any special condition for iOS.
- There's no enable (or showBluetoothSettings) function in the iOS-specific code.

As suggested before, could you please make the iOS version of enable (and showBluetoothSettings) always call success, or handle Bluetooth activation of course, if that's possible?

don, 2015-12-20:
Unfortunately there's no way to enable Bluetooth on iOS. There are some circumstances, like when the app is first run, where iOS will prompt the user to enable Bluetooth, but as a developer you have no control over this.

I don't think it's is the correct behavior for library to return success when the `enable` function isn't available. I should probably make the functions always call failure with a "Not supported" message, but that doesn't help you.

**You can add code to your app to get the behavior you want.**

After `deviceready` is fired, you can overwrite the function only on iOS

    if (cordova.platform === 'ios') {
        ble.enable = function(success, failure) {
            success();
        }
    }

Or you could always overwrite the function and add the conditional logic in the new version of the function. Again make sure you do this after `deviceready`.

    ble.enable = function (success, failure) {
        if (cordova.platform === 'ios') {
            success();
        } else {
            cordova.exec(success, failure, "BLE", "enable", []);      
        }
    }

papasmurph, 2015-12-20:
I got such a message, but only once. I don't know why.

This is how I did it:

    if (device.platform.toLowerCase() === "android") {
        ble.enable(
                function() {
                    bleScan();
                },
                function() {
                    log("This app needs Bluetooth enabled.");
                }
        );
    }
    else {
        bleScan();
    }

Result: https://www.youtube.com/watch?v=ACZ-yJ7PvQ0
(just a demo; what I wanted to test was end-to-end behavior, and MIDI is great for that)

papasmurph, 2015-12-21:
By the way, I suggested a reasonable change to the plugin. I didn't ask for how to circumvent an inconsistency in the plugin.

don, 2015-12-21:
I'm glad you got it working. Cool video.

I heard what you asked for, but I suggested the work around because my implementation would be to always fail on iOS rather than always succeed, which wouldn't help in your case.

papasmurph, 2015-12-21:
True, that wouldn't help me very much.

papasmurph, 2015-12-23:
ble.enable doesn't seem to work on Android 4.4.4 either. Not my device so I couldn't make a deeper investigation, but once I removed ble.enable the app worked.

don, 2015-12-28:
@andersborgabiro I don't have a 4.4.4 device with me. On an LG L16C with Android 4.4.2 it worked. Next time you have a 4.4.4 device, see if you can get a stack trace from `adb logcat` and open a new issue. There maybe some conditional logic required for certain devices or Android versions.

closed 2015-12-28 by don

andersborgabiro, 2015-12-30:
Alrighty.