don/cordova-plugin-ble-central #145: ios can't connect device again after the connection is disconnected

yangheng, 2015-12-28:
The first time the device is connected successfully, the connection is disconnected once again been not connect.You have to restart the Bluetooth device to connect again successfully,

don, 2015-12-28:
How are you disconnecting? Is this a duplicate of #126?

don, 2016-04-15:
I'm thinking that you're seeing an orphaned connection similar to #126.

I had similar problems when running Bluetooth LE apps in [PhoneGap Developer App](http://app.phonegap.com).

My solution was to [add a hook](https://github.com/don/ITP-BluetoothLE/blob/master/phonegap/connect/www/js/disconnector.js#L65) that runs before `window.location.reload` and disconnects the Bluetooth LE connections.

I'm closing as a duplicate of #126

closed 2016-04-15 by don