don/cordova-plugin-ble-central #146: Document error object for connect() failure

RangerMauve, 2015-12-28:
Currently it's not obvious as to what the format for errors that occur during connecting looks like. It'd be useful if there was a quick blurb for what one might expect to see. Mostly, it'd be useful to distinguish between device disconnection and errors during connection. Obviously, one could see the format by actually inspecting the error when it happens, but it'd be useful to have handy without having to go through the trouble of setting up a test to create such a scenario.

RangerMauve, 2015-12-28:
Just looked into the [code for Android](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L181), and it seems that no useful information is being given. Do you think it'd maybe make sense to add some sort of code to the data so that it'd be easier to know exactly what happened and provide a better user experience than just "herp it's gone"

don, 2015-12-29:
Currently there's no error object. You get the peripheral object in the error callback for connect when a disconnection happens. (This is so you can tell which peripheral disconnected, when there are multiple connections.)

In the [case you highlighted](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/Peripheral.java#L181), I don't know what else I can pass back. [onServicesDiscovered](http://developer.android.com/reference/android/bluetooth/BluetoothGattCallback.html#onServicesDiscovered(android.bluetooth.BluetoothGatt,%20int)) doesn't give you a lot. If you find cases where the status indicates useful info, let me know and we can incorporate it into the plugin.

RangerMauve, 2015-12-29:
I guess, it'd be useful to just have some information on what point it errored out at. For example, whether it was connected but then disconnected, or if the service discovery failed. Also it seems that [sometimes](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/BLECentralPlugin.java#L211) the error will actually be a string if the peripheral wasn't previously scanned.

RangerMauve, 2015-12-29:
In my use-case I'd like to record the specific failures for analysis when users complain that their device isn't working as well as make it easier to distinguish what sort of prompt I should give users when things go wrong.

don, 2016-01-11:
For now, you'll need to manage your error callback to determine what failed. Some places give you string error messages, other places particularly on Android might return an error code. It's not ideal in all cases, but not a high priority for me to fix until I'm running across this issues or clients are paying for enhancements. If you have improvements, I'm willing to consider pull requests to make this better.