don/cordova-plugin-ble-central #150: Notifications not always working

Janckk, 2016-01-05:
Hi,

so I'm able to register to notifications by using startNotification. Everything works OK. But when I start fiddling with my external device that I'm connecting to I'm not able to get any response from device anymore. Writing works, but I don't get anything returned back. For example: I'm connected to device, I turn off external device and turn it back on, reconnect to the device and now I can't get any notifications.

I've tried debugging and found out that onCharacteristicChanged function is not being called at all. Although a command with code 10001 (NOTIFY) is being called and NOTIFY is being written to the descriptor.

Is there any advice on this, resetting both devices does not solve the problem! The only way I can solve the problem is by using a chip manufacturer application -> MLDP terminal (source code is attached). When I use it I get response right away. 

[mldpterminal2.zip](https://github.com/don/cordova-plugin-ble-central/files/78565/mldpterminal2.zip)

don, 2016-01-05:
@Janckk if `onCharacteristicChanged` isn't being called, it seems like the notification is not making it to Android, or perhaps startNotification failed to set the descriptor value.

Can you try repeating your test case with a tool like [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en)? Do you see the same behavior?

Janckk, 2016-01-05:
How would I know, I don't get any success or error response when calling ble.startNotification. When I'm testing the "if (gatt.writeDescriptor(descriptor))" it returns true, so i'm assuming it gets written to the descriptor.

I will try repeating it tomorrow (with nRF), will report the results.

Janckk, 2016-01-06:
I am able to get a successful callback from nRF MCP when I click the icon to indicate and it returns indications enabled. 

I haven't been able to write data thought, for now I'm writing data with ble-central and I'm sending two writes, one after another:
stringToBytes(1)
stringToBytes(\n)

I'm not sure how to enter this into nRF MCP. I've tried two values, first (BYTE) 0x31, second (BYTE) 0x0A. 

Any help would be appreciated.

Janckk, 2016-01-06:
I've found this post and it seems that this is exactly my problem: 

http://stackoverflow.com/questions/30088818/ble-gatt-oncharacteristicchanged-not-called-after-subscribing-to-notification/30155119#30155119

I'm not sure how to adjust ble-central plugin for this though.

don, 2016-01-07:
Answering multiple things...

Since this works with nRF MCP, but not the plugin, there's probably something that needs to be updated in my code.

To write multiple bytes with nRF MCP, choose write and choose byte array, enter 310a

To write multiple bytes with the plugin, try something like

     var data = new Uint8Array([0x31, 0x0a]);
     ble.write(data.buffer);

In the [stack overflow post](http://stackoverflow.com/questions/30088818/ble-gatt-oncharacteristicchanged-not-called-after-subscribing-to-notification/30155119#30155119), the user is reusing a UUID for the service and characteristic. Are you doing this? If you are, please update to use unique UUIDs for each attribute.

The client characteristic configuration UUID must be 0x2902. The plugin doesn't have APIs for setting the descriptors yet. The descriptor value is updated when you call `ble.startNotification`. If you disconnect and reconnect, you need to `call bleStartNotification` again. In nRF MCP, when you click the icon to subscribe to notifications, the app write the proper value to 0x2902 for that characteristic.

Yes, Android will cache values. Turning Bluetooth off and on will help. nRF MCP has a "Refresh Services" menu option. I think the *proper* solution is to implement *Service Changed* 0x2a05 on *Generic Attribute Service* 0x1801 indicating to the OS that it needs to re-read the service definition.

Janckk, 2016-01-08:
1. Yes, I'm using different UUID's for service and characteristic.

2. I am calling startNotification on each connect, is it possible that calling it multiple times would cause a problem?

What is the next step?

don, 2016-01-11:
0x2902 is hardcoded because is the UUID for [Client Characteristic Configuration](https://developer.bluetooth.org/gatt/descriptors/Pages/DescriptorViewer.aspx?u=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml). It's the way that notifications and indications are enabled on any characteristic that supports them.

It's possible the DeviceControlActivity.java works all the time for you because they are using both notifications and indications on MLDP_DATA_PRIVATE_CHAR. (See lines 232 and 238).

My plugin handles notifications *or* indications for a particular characteristic but not both. If you really need both for this characteristic, it would require changes to the plugin.

We're mixing multiple issues in this thread. Implementing Service Changed for Generic Attribute is more complicated. It refers to the device caching services and characteristics. Technically  Service Changed should only be required when bonding, but I think iOS and Android uses this to cache values anyway. This is something that needs to be implemented in the peripheral. Are you implementing the peripheral hardware?

Janckk, 2016-01-11:
Hi,

thanks for the input, sadly I'm not implementing the peripheral hardware and it makes my hands a bit tight at the moment. 

I'll try using both notify and indicate and see if it works. But when checking the services with external 
app I only get "indicate" possibility (excue my sloppy terminology).

don, 2016-01-12:
If you're only seeing indicate, that's good. The plugin handles this case.

From the iOS and Android point of view, notify and indicate behave almost exactly the same. The OS handles the confirmation required by indicate, so for either case you just see the data as it arrives.

closed 2016-03-19 by don

shahilpb, 2018-02-13:
Hi JanMesaric/Don,
Hope you are doing well, 
Did you get any solution about **"Android BLE: onCharacteristicChanged never fires"**
I am facing the same issue, Help me, please.