don/cordova-plugin-ble-central #151: How to read from serial.print

laics1984, 2016-01-07:
Hi, i am working on one of the examples, redbearlab connecting to bluno board. I am able to use the read method to get serial.write, but unable to read from serial.print or serial.println. May i know which how to read from the string output via serial.print in the sketch?

don, 2016-01-07:
The examples use a [notification](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/redbearlab/www/js/index.js#L79) to get data written from the redbear lab board on the phone. You could use read, but notification usually better since you don't need to poll, you just get the data when it arrived.

If you can get data from `Serial.write` you should be able to get it from `Serial.print` too. Maybe you need to send a \n character or call `Serial.flush`?

laics1984, 2016-01-08:
Nothing come out from startNotification. txCharacteristic has the same value with rxCharacteristic for bluno board rite? Neither success or failure functions are called.

don, 2016-01-08:
That's ok if TX and RX use the same characteristic. The bluno board will need to support notifications on that characteristic. 

For startNotification, you won't receive the success callback until the bluno sends data. The success callback will be called over and over every time data arrived from the bluno.

Do you have the Bluno BLE service definition? Maybe there's another flow control characteristic or configuration that is necessary?

laics1984, 2016-01-08:
Yup, i have the service definition too. 

here is the sketch from bruno board:

```javascript
unsigned char data;
 void setup() {
  Serial.begin(115200);  //initial the Serial
}
void loop()
{
   if (Serial.available()){
    data = Serial.read();
    Serial.write(data);
    Serial.println();
   
    if (data == 'a'){
      Serial.print("Acknowledgement");
    }
    Serial.println();   //print line feed character
  }
}
```

don, 2016-01-08:
I don't have the service definition or a bluno.

Can you create a repo with your Arduino and Cordova code? I'll try and make another device work like the bleno. 

Also, are you running the latest bluno firmware? The [wiki](http://www.dfrobot.com/wiki/index.php?title=Bluno_SKU:DFR0267) says there's a bug in old versions.

laics1984, 2016-01-09:
Hi don, this is the repo that based on one of the examples https://github.com/laics1984/redbearlab, with the sketch. I can retieve the value using bluetooth ble utility app on ios with service uuid DFB0 and characteristic uuid DFB1.

laics1984, 2016-01-17:
Hi @don , anything wrong with my code? uuid? i hv yet to get it run.. guidance needed. Can you please help? Thanks.

don, 2016-01-26:
At first glance the Cordova code looks fine, since it's basically my example with the UUIDs changed. Unfortunately I can't run it since I don't have a Bluno.

Maybe you need to do something different on the Arduino side? Are you running the latest Bluno firmware?

See if you can get your example working with [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) or [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8). If that doesn't work, talk to the Bluno people to see what they suggest. If you can get the code working with nRF Master Control Panel or LightBlue it should work with my plugin.

closed 2016-04-15 by don

cnadeau, 2016-06-03:
I have some problem getting the bluno to enable notifications on serial port and I think it's a firmware issue:

bluno serial service definition: 
{
     "descriptors": [
       {
         "uuid": "2901"
       }
     ],
     "characteristic": "dfb1",
     "service": "dfb0",
     "properties": [
       "Read",
       "WriteWithoutResponse",
       "Write",
       "Notify"
     ]
   }

According to BLE standard:
https://www.bluetooth.com/specifications/adopted-specifications (core version 4.2, page 2229, table 3.10)

descriptor 2902 MUST be present in order to be able to enable/disable notifications for a service, which is not the case and leads to error:

error while starting notifications for service DFB0, characteristic DFB1: "Set notification failed for 0000dfb1-0000-1000-8000-00805f9b34fb"

@don does it make sense? I can't get any other information from the error itself

don, 2016-06-05:
@cnadeau sounds like a peripheral issue. If the firmware update doesn't fix, create a new github issue, ideally with a test app.

don, 2016-06-06:
@cnadeau It looks like Bluno is still broken with firmware 1.96. The Client Characteristic Configuration descriptor (0x2902) is missing. It's not clear where to file the Bluno issue.

cnadeau, 2016-06-06:
Just saw someone posting the exact same issue here, just replied and we'll see we it leads us!

http://www.dfrobot.com/forum/viewtopic.php?f=18&t=2035&p=9218#p9218

thanks a lot for your reply 👍

AllanSCosta, 2016-07-18:
Any updates on this issue? Like laics above, my code does output Serial.write(data), but can't  get to Serial.print("Acknowledgement"). 
In brief, if I write something to the Serial with the app and then read with the app, it basically returns what I've printed.

don, 2016-07-18:
@AllanSCosta this issue is closed so no updates. I'm not sure what the problem is. It's probably that the read is getting more than. 'a' so the if statement never fires. Try matching "a\n"

Can you create an example with code and instructions and post to github ornbitbucket? If it's ble-central related open a new issue here. If it's all DFrobot related, open a stackoverflow issue or DFrobot forum post and email me a link. I'll take a look.

socca1157, 2017-08-07:
It's been a year, was there ever a solution discovered?

socca1157, 2017-08-23:
Still working on this. 
Example output from serial: "45/60".
I can get 2 (60) out of 5 digits from the serial if I use:

```
this.ble.read(this.current_device.id, 'dfb0', 'dfb1').then( data => {
      var strData = String.fromCharCode.apply(null, new Uint8Array(data));
      alert(strData);
});
```

But haven't figured out how to get the entire serial string. I'm attempting to run this function in a 3 second setInterval loop since the Bluno is unable to subscribe to notifs.

Even stranger, if I rearrange the 5 digits the serial is outputting, it still only sees the same 2 digits (60)......

I would LOVE any assistance here, I'm so close to being done!