don/cordova-plugin-ble-central #152: Problem with reconnecting on ios

Janckk, 2016-01-07:
Hi,

I've connected successfully and everything is working ok, when I disconnect and reconnect right away I get this error. Both connects are fired with the same functionality. 

ProjectTest[1364:699914] -[NSNull isEqualToString:]: unrecognized selector sent to instance 0x38a42588
tion:request:frame:decisionListener: delegate: <NSInvalidArgumentException> -[NSNull isEqualToString:]: unrecognized selector sent to instance 0x38a42588

Would love some assistance, 

Sincerely,
Jan

don, 2016-01-07:
Can you find exactly where this is happening? I expect a null UUID is being passed to [findPeripheralByUUID:uuid](https://github.com/don/cordova-plugin-ble-central/blob/4e1394cb9f5970460c38e7b4470546b33b1f5c2f/src/ios/BLECentralPlugin.m#L500). That method is being called from a bunch of places. Is this happening in [connect](https://github.com/don/cordova-plugin-ble-central/blob/4e1394cb9f5970460c38e7b4470546b33b1f5c2f/src/ios/BLECentralPlugin.m#L57)?

Janckk, 2016-01-08:
Yes, it is either connect or startScan. I first fire startScan and after it finds the device call connect via success callback.

don, 2016-01-11:
When you re-connect, and it fails, can you make sure the UUID is not null in your JavaScript and in [this objective-c code](https://github.com/don/cordova-plugin-ble-central/blob/4e1394cb9f5970460c38e7b4470546b33b1f5c2f/src/ios/BLECentralPlugin.m#L55)?

Janckk, 2016-01-11:
For now I'm changing my workflow to require less reconnects and it is working for me. I can do a bit of more research if you want/need it next week?

I think that the problem was writing while the app wasn't connected, there is 15-20s gap between disconnect and the time when connect works again for me.

don, 2016-01-12:
I'm glad the workflow change helped your app. Other users have reported similar issues with the disconnect delay for some devices. If you find more info and there's a way to improve the plugin please let me know. Re-open this issue or create a new one.

Janckk, 2016-01-14:
Ok so Im still having problem although they are fewer due to the fact that Im making less reconnects.

The problem I get to is when I turn off the peripheral device while Im connected to it. I get the connect fail callback and for now Im ble.disconnect just in case inside of it. But when I turn the peripheral back on, I reconnect to it but write and startNotification actions do nothing.

It helps if I turn off and on bluetooth on my device. Is it possible something is cached?

don, 2016-01-26:
If the failure callback from ble.connect is called, you're disconnected, you shouldn't need to call ble.disconnect.

There are cases where device information gets cached, turning bluetooth on and off fixes this.

If you're not seeing anything happen after a write, it's possible the device is not sending the write response back to iOS. Can you verify if the response if being sent or not?

closed 2016-03-19 by don

callblueday, 2016-08-08:
Hi, @don , I have the same problem when I invoke `ble.startScan()` method. This problem happends when the peripheral device's uuid is null. The method `findPeripheralByUUID` is in `BLECentralPlugin.m`

```
- (CBPeripheral*)findPeripheralByUUID:(NSString*)uuid {

    CBPeripheral *peripheral = nil;

    for (CBPeripheral *p in peripherals) {

        NSString* other = p.identifier.UUIDString;

        if ([uuid isEqualToString:other]) {
            peripheral = p;
            break;
        }
    }
    return peripheral;
}
```

and i fixed the problem like this

```
- (CBPeripheral*)findPeripheralByUUID:(NSString*)uuid {

    if (!uuid || uuid == [NSNull null]) {
        return nil;
    }

    CBPeripheral *peripheral = nil;

    for (CBPeripheral *p in peripherals) {

        NSString* other = p.identifier.UUIDString;

        if ([uuid isEqualToString:other]) {

            peripheral = p;
            break;
        }
    }
    return peripheral;
}
```
I wonder should the uuid be null?

don, 2016-08-08:
The UUID should never be null. Can you reliably reproduce this? What is the peripheral? Which versions of iOS?

Aeva-Mac, 2019-08-26:
i have the same probleme, helps