don/cordova-plugin-ble-central #153: startNotification callbacks

CristuNaranjo, 2016-01-08:
Hi,

when I register a notification for a characteristic, I only have a failure parameter callback, but never in success. What parameters should return failure? The error? Because I receive the new value of the characteristic and I don't know if I should use this, because for now it's working, but in the bad call. Never get inside success callback but I believe that in this success callback should be the new value of the characteristic, no?

CristuNaranjo, 2016-01-11:
Leave here the code which I'm testing...
Always get in failure, but the buffer return the new characteristic data. Can you help me? Need more info?

```javascript
var service_test = '12345678-90ab-cdef-fedc-ba0987654321';
var characteristic_test = "11121314-1516-1718-1910-1a1b1c1d1e1f";
ble.startNotification(deviceId, service_test, characteristic_test,
	function(buffer){
		alert('success');
		console.log(buffer);
		var data = new Uint8Array(buffer);
		console.log(data);
	},
	function(buffer){
		alert('failure');
		console.log(buffer);
		var data = new Uint8Array(buffer);
		console.log(data);
	}
 );
```

don, 2016-01-11:
For `ble.startNotification`, the success callback is called every time your device receives updated data from the notification.

If there's an error, on iOS you'll get the localized error message string rather than a buffer. [code](https://github.com/don/cordova-plugin-ble-central/blob/4e1394cb9f5970460c38e7b4470546b33b1f5c2f/src/ios/BLECentralPlugin.m#L457). For Android you get the error code.

The most likely error is calling startNotification on a characteristic that doesn't support notifications.

CristuNaranjo, 2016-02-08:
I solved the problem. Was a mistake with the name of parameters in a 'factory'. I'm developing in angularJS and i'm a newbie! Thanks for your time, you can close this thread!

closed 2016-03-04 by don