don/cordova-plugin-ble-central #154: Error Writing to Characteristic for Verification

mrAddy, 2016-01-08:
Hiya.  Using this plugin with v.bttn and i have 30 seconds after connect to send a verification number, but it keeps failing.  Using this code:

` ble.writeWithoutResponse(bttnid, "fffffff0-00f7-4000-b000-000000000000", "fffffff5-00f7-4000-b000-000000000000", data.buffer,
  function() {
    console.log('write success 2');
  },
  function(reason) {
    console.log('BLE fails to write 2 ' + reason);
  }`

The error is,

BLE fails to write 2 Characteristic fffffff5-00f7-4000-b000-000000000000 not found.

But its in the list when it connects ?

[image]

Not sure what i am doing which is wrong ?

Thanks in advance.

don, 2016-01-08:
@mrAddy based on your screenshot `fffffff5-00f7-4000-b000-000000000000` supports write, but not writeWithoutResponse.

    ble.write(bttnid, 
                  "fffffff0-00f7-4000-b000-000000000000", 
                  "fffffff5-00f7-4000-b000-000000000000", 
                  data.buffer,
                  function() { console.log('write success 2'); }, 
                  function(reason) { console.log('BLE fails to write 2 ' + reason); 
    }

mrAddy, 2016-01-08:
Okay, i have tied it without, but then i get error 13, which is a generic error i believe ?

mrAddy, 2016-01-08:
Do I have this correct, there are 36 Characteristics on the device, the one according to the documentation is located under object 29.

So is this code correct to access that, and write to it:

`   console.log(dev.characteristics);
        var data = new Uint8Array(36);
        data[29] = "80:BE:F5:AC:FF";
        //console.log(dev.characteristics[29]);
        ble.write(bttnid, "fffffff0-00f7-4000-b000-000000000000", "fffffff5-00f7-4000-b000-000000000000", data.buffer,
        function() {
            console.log('write success 2');
        },
        function(reason) {
            console.log('BLE fails to write 2 ' + reason);
        });
`

Or is my understanding wrong ?

don, 2016-01-08:
I think error 13 (0xd) means the value is too long for the characteristic [GATT_INVALID_ATTRIBUTE_LENGTH](http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#GATT_INVALID_ATTRIBUTE_LENGTH). Try sending a smaller ArrayBuffer.

The UUID gets you a handle to the proper characteristic. You should be able to set a value directly on that characteristic. Without knowing more, I'd do something like 

    var data = new Uint8Array([0x80, 0xbe, 0xf5, 0xff, 0xff]);

To get the MAC address into an typed array.

Do you have a link to the documentation?

mrAddy, 2016-01-08:
There is no link, but here is all they have:

I am not sure how to convert the UUID to a smaller number.  If you look on the docs it has 2901 or F5FF is given for this Characteristic, but it gives a different long UUID.

[v_bttn dev.pdf](https://github.com/don/cordova-plugin-ble-central/files/83257/v_bttn.dev.pdf)

Thank you for your help Dom :)

don, 2016-01-08:
The UUID isn't too long, the data you're passing to the characteristic is too long.

The verification key looked like a mac address, but based on the vague docs, maybe it's a String?

Try this...

    // ASCII only
    function stringToBytes(string) {
        var array = new Uint8Array(string.length);
        for (var i = 0, l = string.length; i < l; i++) {
            array[i] = string.charCodeAt(i);
        }
        return array.buffer;
    }

    var verificationKey = stringToBytes('80:BE:F5:AC:FF');

    ble.write(bttnid, 
        'fffffff0-00f7-4000-b000-000000000000', // v.bttn service
        'fffffff5-00f7-4000-b000-000000000000', // verification characteristic
        verificationKey,
        function() { console.log('wrote verification key'); }, 
        function(reason) { console.log('Error sending verification key ' + reason); 
    }

mrAddy, 2016-01-08:
Great, but I still get the error 13:

Error sending verification key 13 

I have tried to contact their support, but nothing in 3 days :(

don, 2016-01-08:
How about if you send the verification data as bytes?

    var data = new Uint8Array([0x80, 0xbe, 0xf5, 0xac, 0xff]);

don, 2016-01-08:
Another option is to see if you can get the verification code to work using [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) or [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8). Then once you know what works, translate that to code. Try things like reversing the byte order on the verification code etc.

mrAddy, 2016-01-08:
I think the verification variable is coming back empty.

if I put a `console.log(verification)`  I have in console {}  but i thought i would get an object?

don, 2016-01-08:
ArrayBuffers look empty if you log them

    > var data = stringToBytes('80:BE:F5:AC:FF');
    undefined
    > data
    ArrayBuffer {}

You need to create a typed array if you want to see the values

    > new Uint8Array(data)
    Uint8Array {
      '0': 56,
      '1': 48,
      '2': 58,
      '3': 66,
      '4': 69,
      '5': 58,
      '6': 70,
      '7': 53,
      '8': 58,
      '9': 65,
      '10': 67,
      '11': 58,
      '12': 70,
      '13': 70 }

don, 2016-01-08:
You need to pass the verification key as bytes. Looking at the screenshot on [stackoverflow](http://stackoverflow.com/questions/34683435/write-verification-to-cordova-ble-error-using-javascript), the table says the characteristic is 5 bytes.

    var verificationKey = new Uint8Array([0x80, 0xbe, 0xf5, 0xac, 0xff]);

    ble.write(bttnid, 
        'fffffff0-00f7-4000-b000-000000000000', // v.bttn service
        'fffffff5-00f7-4000-b000-000000000000', // verification characteristic
        verificationKey.buffer,
        function() { console.log('wrote verification key'); }, 
        function(reason) { console.log('Error sending verification key ' + reason); 
    }

mrAddy, 2016-01-08:
You are a star Dom.  It worked.  Thank you so much.

I am guessing now its connected, I should be able to use the Plugin Docs for the rest ?  Like button presses, and listeners ?

Also, do you have a contribute pot ?

Thanks again.

Andrew

don, 2016-01-08:
Great news! To go further check out the [docs](https://github.com/don/cordova-plugin-ble-central) and [examples](https://github.com/don/cordova-plugin-ble-central/tree/master/examples). 

If you'd like more detail, I have a new book [Make:Bluetooth](http://shop.oreilly.com/product/0636920031932.do) that has lots of code and explanation. If you have a Barnes and Noble near you they might have it in stock.

Other than that, can you upvote my stack overflow answer and mark it as accepted.

closed 2016-01-08 by don

mrAddy, 2016-01-08:
Just to confirm something.  I seem to keep getting Invalid long.   So each time I try to startNotifications.  If I get have long characteristic (0xFFFFFFF4-00F7-4000-B000- 000000000000) and its too long, how do I convert it to a smaller var, like in your example on startNotification ?   Do I need to do the same thing and convert it ?

don, 2016-01-08:
You should treat the UUIDs in your code as Strings rather than numbers.
0xFFFFFFF4-00F7-4000-B000-000000000000 should be 'FFFFFFF4-00F7-4000-B000-000000000000'

Some charateristics and descriptors may use 16 bit UUIDs like 0x2902. These should also be strings '2902'.