don/cordova-plugin-ble-central #155: Android ble.connect() function quesstions

DoFast, 2016-01-12:
[image]

SOS , when I try to connect device always failed

don, 2016-01-12:
There's not enough information here for me to help you.

DoFast, 2016-01-13:
[image]

In Android 22 alway connect error , the same as BluetoothLE - Rand Dusing's BLE plugin for Cordova
But in Ios is ok

don, 2016-01-13:
What device are you connecting to?

Can you duplicate this on with different phones? Or does this only happen on one phone?

Can you connect with [nRF Master Contol Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en)?

DoFast, 2016-01-13:
[image]
[image]
I use the car OBD device.
And I have three differernt phones all have question

DoFast, 2016-01-13:
nRF Master Contol Panel Always stay connecting....

DoFast, 2016-01-13:
I change function  gatt = device.connectGatt(activity, false, this); to true 
In one of my phone android 5.0.1  is connected ,but the other still failed   
[image]

DoFast, 2016-01-14:
Now  when i connect decvice 
always  Log   "onClientConnectionState() - status=133 clientIf=5 device=xxxxxxx"

closed 2017-08-22 by don