don/cordova-plugin-ble-central #156: if the peripheral is accidentally disconnected, will automatically reconnect?

BIGN0, 2016-01-12:


don, 2016-01-12:
Your application needs to handle the reconnect. The [failure callback for connect](https://github.com/don/cordova-plugin-ble-central#connect) will be called when the peripheral unexpectedly disconnects. Even if the success callback has been called, the failure callback can be called at some point in the future.

    var onConnect = function(peripheral) {
      // the connection was successful
      // save the peripheral.id
      // discover services etc
    }

    var onDisconnect = function(peripheral) {
      // the peripheral unexpectedly disconnected 
      // add your reconnect logic here
    }

    ble.connect(deviceId, onConnect, onDisconnect);

BIGN0, 2016-01-13:
var interval;

ble.connect(deviceId, onConnect, onDisconnect);

var onConnect = function(peripheral) {
    clearInterval(interval )
    //...
}

var onDisconnect = function(peripheral) {
    interval = setInterval(function(){
        ble.connect(deviceId, onConnect, onDisconnect);
    },5000)
}

Do you think it would be okay? 5 seconds to make a re connection,if the peripheral is accidentally disconnected

don, 2016-01-15:
Some devices will need longer after they disconnect. You might add a decaying timer so each reconnection attempt takes a bit longer 5 seconds, 10 seconds, 20 seconds etc. Rather than trying to reconnect forever, you probably want to have it error out at some point.

closed 2016-03-19 by don