don/cordova-plugin-ble-central #157: Peripheral.id on IOS is not the same on different device

flam182, 2016-01-15:
Hi,

When i scan a beacon (RADBeacon X2) on IOS devices differently , I find the UUID ( peripheral.id ) different.

On Android I still have the MAC address of the tag in " peripheral.id "

How on IOS, i can identify a beacon ?

don, 2016-01-15:
On iOS the UUID is generated and hides the MAC address. Different iOS devices scanning the same peripheral generate different UUIDs for the same peripheral. This is done intentionally Apple, there's not much you can do about it. If you control the peripheral side, the peripheral can have a service that give the phone the MAC address or some other unique identifier.

For iBeacons, the identifying information is in the advertising info, not the peripheral's id. You can parse the beacon data from kCBAdvDataManufacturerData key in the [advertising data](https://github.com/don/cordova-plugin-ble-central#advertising-data) or try a different [plugin that specifically handles beacons](https://github.com/petermetz/cordova-plugin-ibeacon).

flam182, 2016-01-26:
Thanks @don 

It's working with the plugin who handles beacons.

closed 2016-01-26 by don