don/cordova-plugin-ble-central #159: Using ble.read to get battery level

mrAddy, 2016-01-17:
I am not sure if the plug-in is designed to read this, because I keep getting a odd value back, and from reading the Bluetooth Docs (https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.battery_service.xml) suggests I should get a value that represents 0 to 100 as a percentage.

But what I get back is {"0" : 9}

Which makes me think the plugin isn't able to achieve this ?

`function readBttnBattery() {
    var devid = localStorage.getItem('vBttn');
    ble.read(devid, '180f',
             '2a19', 
             function(t) {
        var data = new Uint8Array(t)
        console.log('button Battery:');
        console.log(JSON.stringify(data[0]));
    }, function(f) {
        console.log(f);
    });
}`

Now, the other day I was getting {"0" : 57}  but now i get just "9" ?

don, 2016-01-17:
The plugin reads values from characteristics and is definitely designed to read things like the battery level. The values you're getting are in the proper range. Why do you think it's incorrect? Seem likely the battery level dropped from 57% the other day to 9% today.

Can you compare the values your reading in your code with the same characteristics read with [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) or [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8)?

Try replacing the battery and see if you get different values.

mrAddy, 2016-01-17:
I have had a look at the example, but I cannot understand a new device, with a new battery, would jump that quickly ?.  But it does say in the docs that keeping fall detection it will vastly reduce battery, but its been a week, so, that certainly is vastly when by design its meant to last a year.

mrAddy, 2016-01-17:
I think I have enabled notifications for the battery, but not sure its working, or even how to check.  I guess if, at some point, it drops to 8 i will see ?

mrAddy, 2016-01-17:
Yeah, you are correct, I looked on that nRF you showed me before, and its 9%.  Thats an eye opener!

don, 2016-01-17:
In my experience it's easy to kill batteries quickly, especially during development.

If the device supports notifications for battery, that's better than reading. With the notification, the device will tell you whenever the battery level change. If you're polling to read the battery value, this will help wear out the battery.

You could look at my [battery level example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/battery) and try [uncommenting the line that handles starts notifications](https://github.com/don/cordova-plugin-ble-central/blob/master/examples/battery/www/js/index.js#L65).

mrAddy, 2016-01-17:
I have used that line and I can see from nRF notifications are enabled.  So I guess it's turned on, or able. But thank you :)

mrAddy, 2016-01-17:
Don, do you use any signal bars with any of your examples.  I have looked, as I wanted it the same as the battery, but all i can find is you need to scan it again ?

closed 2016-01-20 by mrAddy