don/cordova-plugin-ble-central #161: Plugin fails to compile under cordova-ios@4.0.1

macdonst, 2016-01-19:
I updated a project to use cordova-ios version 4.0.1 and the plugin no longer compiles.

```
** BUILD FAILED **

The following build commands failed:
	CompileC build/PhoneGap.build/Debug-iphonesimulator/PhoneGap.build/Objects-normal/i386/CBPeripheral+Extensions.o PhoneGap/Plugins/cordova-plugin-ble-central/BLECentralPlugin/CBPeripheral+Extensions.m normal i386 objective-c com.apple.compilers.llvm.clang.1_0.compiler
(1 failure)

Error code 65 for command: xcodebuild with args: -xcconfig,/Users/smacdona/code/phonegap-app-developer/platforms/ios/cordova/build-debug.xcconfig,-project,PhoneGap.xcodeproj,ARCHS=i386,-target,PhoneGap,-configuration,Debug,-sdk,iphonesimulator,build,VALID_ARCHS=i386,CONFIGURATION_BUILD_DIR=/Users/smacdona/code/phonegap-app-developer/platforms/ios/build/emulator,SHARED_PRECOMPS_DIR=/Users/smacdona/code/phonegap-app-developer/platforms/ios/build/sharedpch
```

don, 2016-01-19:
@macdonst I replaced the api call that was removed. Can you try again with master? This worked good with my extremely limited testing.

macdonst, 2016-01-19:
@don yeah, that seems to fix the compile error. Lemme know if you are bumping the NPM package so I can pin to a new version.

closed 2016-01-19 by macdonst

don, 2016-01-19:
pushed version 1.0.4 to npm

macdonst, 2016-01-19:
@don thx