don/cordova-plugin-ble-central #162: ios not discover phone with android

langowskiadam, 2016-01-20:
Hello,
since few days I was trying to figure it out and I failed. When I use scan command it nor discover  my android phone, but in my iPad bluetooth settings it is discovered (see screen bellow). I need only to get uuid, I don't want to connect/write etc -  generally I only want to discover devices, which are in bluetooth range.

Please help.

Kind regards.

[image]

francis-cy, 2016-01-23:
I encountered the same problem as well.  ble.scan is unable to detect an android or iOS phone.  Can anyone help?

don, 2016-01-26:
In order for this plugin to discover another phone, that phone needs to be *advertising*. The easiest way to do this is to have the target phone advertising as a Bluetooth Low Energy peripheral. It's also possible to advertise without actually offering services.

francis-cy, 2016-01-27:
Forgive me for my ignorance. Is it just a phone setting to get the phone (both Android and iOS) to advertise as a peripheral, or an app is necessary to achieve that?  If an app need to be written, are there Cordova plugins available, particularly for Android phones?  I think we can do this with native Android code, but I am using Cordova.  Thanks.

don, 2016-01-27:
For now I think you need native code to make iOS and Android advertise Bluetooth services. I don't know of any Cordova plugins to do this. I've been working on cordova-plugin-ble-peripheral but it's not complete.

francis-cy, 2016-01-31:
Thank you.  Hope to see your plugin soon.

closed 2016-02-12 by don