don/cordova-plugin-ble-central #163: Sent Error Response - Attribute Not Found, Handle: 0x0001

furnaceX, 2016-01-21:
Thanks for this plugin.  Very helpful!

I'm trying to replicate an app that connects to ble device.  With the pre-existing app, the full functionality work.  With my app, I can't seem to get the button click to work.  I captured the bluetooth log for each and compared them wireshark.  I notice that during the connection process, my app (using ble-central plugin) gets an error and doesn't seem to read one of the characteristics.  (Though I don't seem to be able to read the char with any available ble scanners, either.)

The left hand column is the working app.  The left hand column is my app using ble-central.

[image]

the error is: Sent Error Response - Attribute Not Found, Handle: 0x0001

It looks like the working app gets data back on a handle at 0x0028.  Later in the working app, that handle is used for a callback on a button click.

Any ideas?

furnaceX, 2016-01-21:
The contents of the packet (sent by the ble device) just prior to the error:  
```
No.     Time        Source                Destination           Protocol Length Info
130     38.175706   remote ()             localhost ()          ATT      18     Rcvd Find By Type Value Request, GATT Primary Service Declaration, Handles: 0x0001..0xffff

Frame 130: 18 bytes on wire (144 bits), 18 bytes captured (144 bits)
    Encapsulation type: Bluetooth H4 with linux header (99)
    Arrival Time: Jan 20, 2016 17:00:50.514920000 MST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1453334450.514920000 seconds
    [Time delta from previous captured frame: 0.043735000 seconds]
    [Time delta from previous displayed frame: 0.043735000 seconds]
    [Time since reference or first frame: 38.175706000 seconds]
    Frame Number: 130
    Frame Length: 18 bytes (144 bits)
    Capture Length: 18 bytes (144 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: hci_h4:bthci_acl:btl2cap:btatt]
    Point-to-Point Direction: Received (1)
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
    .... 0000 0000 1011 = Connection Handle: 0x000b
    ..10 .... .... .... = PB Flag: First Automatically Flushable Packet (2)
    00.. .... .... .... = BC Flag: Point-To-Point (0)
    Data Total Length: 13
Bluetooth L2CAP Protocol
    Length: 9
    CID: Attribute Protocol (0x0004)
Bluetooth Attribute Protocol
    Opcode: Find By Type Value Request (0x06)
    Starting Handle: 0x0001
    Ending Handle: 0xffff
    UUID: GATT Primary Service Declaration (0x2800)
    Value: 0218

0000  02 0b 20 0d 00 09 00 04 00 06 01 00 ff ff 00 28   .. ............(
0010  02 18                                             ..
```

don, 2016-01-22:
Are you running on iOS or Android? 
What OS version? 
What hardware?

You're seeing the error after calling `ble.connect`?

Is it possible the other app initializes something that makes the characteristic available?

furnaceX, 2016-01-22:
Android 5.1
Moto X (original) 

Yes, I see the error in the bluetooth capture after ```calling ble.connect```

In the bluetooth capture, prior to the error, I all the communications to the ble device are the same.  Is it possible that something on the android side (that isn't communicated to the device) isn't initialized?

don, 2016-01-26:
I'm not sure why you're seeing a difference with my library vs the manufacturer's app. Maybe they're passing different arguments to [BluetoothDevice.connectGatt](http://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback))

That error is pretty low level. To debug stuff like this, I usually try and write a trivial native app to isolate the issue.

closed 2016-08-18 by don