don/cordova-plugin-ble-central #165: Will specifying a service for scan speed up discovery

RangerMauve, 2016-01-27:
Currently I'm facing issues where in a crowded environment it's taking a long time to find devices that are actually relevant to my application. I know that these devices have services that are unique to them. Would specifying that service speed up discovery of those devices, or is it just a filter that's performed on the plugin's side?

My primary platform is Android 4.2

don, 2016-01-27:
You can try filtering scan results by uuid, but on early android versions it doesn't work very well. I think the peripheral hardware, phone, and android version will effect the scan time.

RangerMauve, 2016-01-27:
So, how does the list of services passed into `scan` affect things? Is it passed to the native APIs, or just used in the plugin itself?

RangerMauve, 2016-01-27:
[Looking at the android code](https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/BLECentralPlugin.java#L312), the service UUIDs actually do get passed lower down into the Bluetooth API so I'll test out to see if this helps decrease the time until discovery. Thanks for the help, @don!

closed 2016-01-27 by RangerMauve

Juanelm, 2016-08-13:
RangerMauve, can you comment if you had success on speeding up the scan by passing the service UUIDs?

RangerMauve, 2016-08-15:
@Juanelm Couldn't get it to work. Passing in UUIDs is apparently buggy on a lot of Android devices and we couldn't get it to work on ours.