don/cordova-plugin-ble-central #167: Integrate Bluetooth-Crash-Resolver

RangerMauve, 2016-01-29:
Android's scanning [is known to be buggy](http://www.androidpolice.com/2014/05/20/bug-watch-bluetooth-will-begin-crashing-after-encountering-too-many-ble-devices-affects-kitkat-4-4-and-jelly-bean-4-3/) and can reach states where bluetooth becomes unusable unless you do a factory reset. I've come across this issue several times in the field and it's pretty frustrating. It'd be handy if this plugin made use of [bluetooth-crash-resolver](https://github.com/RadiusNetworks/bluetooth-crash-resolver) so that it could help alleviate this issue. The drawback from using it would be a slightly higher battery use.

don, 2016-04-26:
I think it would be better to have a Bluetooth crash resolver as a separate plugin.

RangerMauve, 2016-04-26:
The thing that'd make that a bit more annoying is that you need to pass the scan callback into `bluetoothCrashResolver.notifyScannedDevice()`, so it'd be a pain to have a separate cordova plugin that somehow got a reference to the `cordova-plugin-ble-central` plugin instance, which doesn't sound like good practice

don, 2016-04-26:
That would be annoying, I'll revisit this again later.

RangerMauve, 2016-04-27:
A thing to consider is that Google apparently patched this for Android 6.0.1, so it won't be an issue in a few years.

closed 2018-06-07 by don