don/cordova-plugin-ble-central #169: Getting subscribed notifications on iOS not working in background

tomasbedrich, 2016-01-30:
When the app is in background mode, the subscribed notifications will stop working after a while. It may be related to [UIBackgroundModes](https://developer.apple.com/library/ios/documentation/General/Reference/InfoPlistKeyReference/Articles/iPhoneOSKeys.html#//apple_ref/doc/plist/info/UIBackgroundModes) option in `Info.plist`, [which is required by Apple](https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html).

The alternative `cordova-plugin-blootoothle` handles this [somewhere in the code](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/4400f6e99aa67b3a2bd78d8e5c18101a7e3405a9/plugin.xml#L34) (not verified, if it's working).

tomasbedrich, 2016-02-05:
I have tested the version with `UIBackgroundModes` option set to `bluetooth-central` and it seemed to work well.

don, 2016-03-19:
I'm a bit concerned about having all apps set UIBackgroundModes. I'm not sure if this can be configurable or it would be better to do with the Cordova Custom Config plugin, https://www.npmjs.com/package/cordova-custom-config

tomasbedrich, 2016-03-19:
Can you please describe why are you concerned about that?

commit 38ebd12b (2016-04-26): iOS background notifications

fixes #169
fixes #216
closed 2016-04-26 by don