don/cordova-plugin-ble-central #170: ble.isEnabled not supported on iOS ?

zabojad, 2016-02-01:
Hi !

The failure callback of the [isEnabled](https://github.com/don/cordova-plugin-ble-central#isenabled) method is always called on iOS (tested on iOS 9.2.1). Isn't this method supported on iOS ?

Thanks !
Tom

zabojad, 2016-02-01:
OK, I just realized that it's not the failure callback that is called. It's the success callback without any parameter like on Android (a boolean representing the connection state).

So it's working... but we have a slightly different API for Android and for iOS... I think the success callback of this method should have always true in its args... What do you think about that ?

don, 2016-02-02:
For [ble.isEnabled](https://github.com/don/cordova-plugin-ble-central#isenabled) both platforms *should* be calling the success callback when it's enabled and the failure callback when it's disabled. I don't think there are any arguments to the callback. I just ignore them. [example](https://github.com/don/cordova-plugin-ble-central#quick-example-6)

zabojad, 2016-02-17:
So shouldn't we update the README as it is stating: 
> success: Success callback function that is invoked with **a boolean for connected status**.

And I checked, this boolean is passed on Android...

don, 2016-02-17:
I think the Boolean for android is a mistake or holdover from old code.  I'll look into this soon. There's a pull request #136 that sends callbacks on Bluetooth state change that may supercede this function.

commit 4c832d2b (2016-03-19): Update documentation for isEnabled

Fixes #170
closed 2016-03-19 by don