don/cordova-plugin-ble-central #173: android can not find other ble devices

dayukiller, 2016-02-02:
hi，i'm using the plugin to write a ble app;but when i use this plugin, i can not find the other devices through bluetooth;here is my code:

`onDeviceReady: function() {
        console.log('hello world');

        ble.scan([], 20, function(device) {
            console.log('success');
        }, function() {
            console.log('fail');
        });
    }`
it looks very simple; but i can not find the other devices(open the bluetooth already)
seek for help, 3q!

don, 2016-02-10:
Are you sure the other devices near you are advertising Bluetooth LE serivces? Use another tool like [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) or [Light Blue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8) and ensure your phone can find devices. You should see similar results when scanning from Cordova.

closed 2016-02-25 by don

dayukiller, 2016-02-25:
I can not use this plugin to find my cellphone, but i can find my devices using bluetooth like my bluetooth scales. If i use the plugin uncorrectly to find my cellphones?Or it can only find the devices not cellphones?

don, 2016-02-25:
@dayukiller the plugin will find phones *if* they are advertising Bluetooth LE services. You can use LightBlue Explorer to create virtual peripherals on iOS devices and nRF Master Control Panel to create virtual peripherals on newer Android devices.