don/cordova-plugin-ble-central #174: Question: iOS prompt for Bluetooth on every app start

H0rst, 2016-02-03:
Hello,

in my App the user can turn Bluetooth on and off. But on iOS the App asks at every startup for activation of Bluetooth in system settings, when Bluetooth is off at the device.

Is this correct, that the App asks at every start? I read in another issue it should ask only at first start.

Can i do something to suppress this message, when Bluetooth is off in my App?

best regards
Horst

H0rst, 2016-02-04:
I have changed line 40 in `BLECentralPlugin.m` to
```
if (NSFoundationVersionNumber > NSFoundationVersionNumber_iOS_6_1) {
	manager = [[CBCentralManager alloc] initWithDelegate:self queue:nil options:@{CBCentralManagerOptionShowPowerAlertKey:[NSNumber numberWithBool:NO]}];	
} else {
    manager = [[CBCentralManager alloc] initWithDelegate:self queue:nil];
}
```

`CBCentralManagerOptionShowPowerAlertKey` will surpress the prompt, but it is available at iOS7 and higher.

Maybe there should a option in the Plugin to use this behavior. For me it works.

don, 2016-02-12:
This should probably be an option.

zabojad, 2017-03-13:
It would be very handy it this were exposed as a parameter in the config.xml. Would that be feasible for the next plugin release ?

don, 2018-06-07:
Fixed by #584

closed 2018-06-07 by don