don/cordova-plugin-ble-central #175: ble connection issue on Galaxy S6

cajay01, 2016-02-05:
I am using ble centarl plugin in cordova ionic framework for connecting custom profile ble 4.0 device. I got success for almost every android smartphone which supports ble,but for Galxy S6 it is not working.Is there any specific reason it is not connecting to ble  on Galaxy S6?or am I missing something?

don, 2016-02-10:
Are you filtering scan results by UUID? This is broken on some versions of Android and some devices. Scanning for all devices by passing an empty array.

Sometimes the Bluetooth stack crashes. Disable Bluetooth and wifi, then re-enable, or restart your phone.

If neither of those work, check `adb logcat` for error messages or stack traces.

cajay01, 2016-02-12:
In scan Results I can see the custom ble device but when I am trying to connect with it,it fails.I need to try multiple times to connect it(7-8 times).I am sharing logs with you where DA:3F:0A:6E:1B:A3 is my device mac address

02-10 16:09:39.451  13911-13924/? D/BluetoothGatt﹕ onClientRegistered() - status=0 clientIf=9
02-10 16:09:39.451    5876-5886/? D/BtGatt.GattService﹕ clientConnect() - address=DA:3F:0A:6E:1B:A3, isDirect=true set own addr = false own addr type:0
02-10 16:09:39.451    5876-5935/? D/BtGatt.btif﹕ btif_get_device_type: Device [da:3f:0a:6e:1b:a3] type 2, addr. type 1
02-10 16:09:39.551  13911-13911/? D/CordovaLog﹕ file:///android_asset/www/js/controllers/scanDevicesCtrl.js: Line 122 : stopScanSuccess
02-10 16:09:39.551  13911-13911/? I/chromium﹕ [INFO:CONSOLE(122)] "stopScanSuccess", source: file:///android_asset/www/js/controllers/scanDevicesCtrl.js (122)
02-10 16:09:40.191    5876-5933/? D/btif_config_util﹕ btif_config_save_file(L188): in file name:/data/misc/bluedroid/bt_config.new
02-10 16:09:40.281    3495-3784/? D/InputReader﹕ Input event(4): value=1 when=1370260569000
02-10 16:09:40.281    3495-3784/? D/InputReader﹕ Input event(4): value=1 when=1370260569000
02-10 16:09:40.281    3495-3784/? I/InputReader﹕ Touch event's action is 0x0 (deviceType=0) [pCnt=1, s=0.117 ] when=1370260569000
02-10 16:09:40.281    3495-3784/? D/InputReader﹕ lastThreadEndTime = 1369352683193, currentThreadStartTime = 1369352688193, diff = 0
02-10 16:09:40.281    3495-3783/? I/InputDispatcher﹕ Delivering touch to (3899): action: 0x4, toolType: 1
02-10 16:09:40.281    3495-3783/? I/InputDispatcher﹕ Delivering touch to (13911): action: 0x0, toolType: 1
02-10 16:09:40.281  13911-13911/? D/ViewRootImpl﹕ ViewPostImeInputStage ACTION_DOWN
02-10 16:09:40.321    3495-3784/? D/InputReader﹕ Input event(4): value=0 when=1370306789000
02-10 16:09:40.321    3495-3784/? D/InputReader﹕ Input event(4): value=0 when=1370306789000
02-10 16:09:40.321    3495-3784/? I/InputReader﹕ Touch event's action is 0x1 (deviceType=0) [pCnt=1, s=] when=1370306789000
02-10 16:09:40.321    3495-3783/? I/InputDispatcher﹕ Delivering touch to (13911): action: 0x1, toolType: 1
02-10 16:09:40.361  13911-13911/? D/cr_Ime﹕ [ImeAdapter.java:213] updateKeyboardVisibility: type [0], flags [0], show [true]
02-10 16:09:40.361  13911-13911/? D/cr_Ime﹕ [AdapterInputConnection.java:178] updateState [] [0 0] [-1 -1] [true]
02-10 16:09:44.451    5876-5937/? W/bt-btif﹕ bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x0100
02-10 16:09:44.451    5876-5937/? W/bt-btif﹕ bta_gattc_conn_cback() - cif=5 connected=0 conn_id=5 reason=0x0100
02-10 16:09:44.451    5876-5937/? W/bt-btif﹕ bta_gattc_conn_cback() - cif=6 connected=0 conn_id=6 reason=0x0100
02-10 16:09:44.451    5876-5937/? W/bt-btif﹕ bta_gattc_conn_cback() - cif=7 connected=0 conn_id=7 reason=0x0100
02-10 16:09:44.451    5876-5937/? W/bt-btif﹕ bta_gattc_conn_cback() - cif=8 connected=0 conn_id=8 reason=0x0100
02-10 16:09:44.451    5876-5937/? W/bt-btif﹕ bta_gattc_conn_cback() - cif=9 connected=0 conn_id=9 reason=0x0100
02-10 16:09:44.451    5876-5935/? D/BtGatt.GattService﹕ onConnected() - clientIf=9, connId=0, address=DA:3F:0A:6E:1B:A3
02-10 16:09:44.451  13911-13924/? D/BluetoothGatt﹕ onClientConnectionState() - status=133 clientIf=9 device=DA:3F:0A:6E:1B:A3
02-10 16:09:44.471  13911-13911/? D/CordovaLog﹕ file:///android_asset/www/js/controllers/scanDevicesCtrl.js: Line 81 : ZenRelaxie Error Disconnected
02-10 16:09:44.471  13911-13911/? I/chromium﹕ [INFO:CONSOLE(81)] "ZenRelaxie Error Disconnected", source: file:///android_asset/www/js/controllers/scanDevicesCtrl.js (81)
02-10 16:09:44.491  13911-13911/? D/CordovaLog﹕ file:///android_asset/www/js/controllers/headerController.js: Line 157 : Connection  is read undefined
02-10 16:09:44.491  13911-13911/? I/chromium﹕ [INFO:CONSOLE(157)] "Connection  is read undefined", source: file:///android_asset/www/js/controllers/headerController.js (157)
02-10 16:09:45.031    3495-5980/? D/SSRM:n﹕ SIOP:: AP = 330, PST = 318 (W:10), CP = 258, CUR = 156, LCD = 44

Aman18493, 2016-02-29:
i am having the same problem,did u find any solution for it

cajay01, 2016-03-08:
Hey Aman we minimized the broadcast time of our ble device later on start works fine with S6

closed 2016-03-09 by cajay01