don/cordova-plugin-ble-central #180: Connect() with pincode

madstt, 2016-02-10:
I have a peripheral that requires a pincode to connect. Is this even possible? Is it on the roadmap?

RangerMauve, 2016-02-10:
Currently, it seems that devices that require pairing should be paired through the native bluetooth UI, and then they can magically connect somehow. Try fiddling with that until @don responds.

don, 2016-02-10:
Ideally the OS will prompt for a pin when connecting or when accessing a characteristic with permissions requiring additional encryption. (This might work better on iOS, but I'm pretty sure it works on Android too.)

madstt, 2016-02-12:
Ok, thanks. I'll have to verify that options...

closed 2016-02-12 by madstt

hervethouzard, 2016-10-04:
Hello,

Has someone verified that it is automatically managed by the OS?