don/cordova-plugin-ble-central #182: Support Android 23

don, 2016-02-12:
Modify the plugin for new Android 23 permissions.

Until then to use the plugin with Cordova 6.0 add the following to config.xml

        <preference name="android-targetSdkVersion" value="22"/>

tomasbedrich, 2016-02-14:
I would like to vote up for solving this ASAP. The plugin does nothing (scanning doesn't work) without this change in `config.xml`.

don, 2016-04-15:
Fixed in cc655b418db2d42149e3936359fc3d7467c8e749

Released in version 1.0.5

closed 2016-04-15 by don

tomasbedrich, 2016-04-15:
Good job @don! ;)

Gonoshift, 2016-06-07:
Hello @don 
I'm using v 1.1.0 of the plugin with Android 6.0.1 (Cyanogenmod) and I still had to make it work I had to modify the  AndroidManifest.xml from:
`<uses-sdk android:minSdkVersion="16" android:targetSdkVersion="23" />`
to:
`<uses-sdk android:minSdkVersion="16" android:targetSdkVersion="22" />`

So the issue is not solved..

Anyway, thanks for your hard work!