don/cordova-plugin-ble-central #185: ble.write fails once then callbacks don't ever fire

azoerb, 2016-02-23:
I have some code that calls ble.write every 10 seconds after the connection has been successfully made.  I'm double checking that the connection is still active (it is) right before writing.  I get one failure callback the first time i call ble.write, and then nothing after that, even though that function is executing every 10 seconds.  I'm connecting to an Adafruit Feather 32u4 bluetooth LE board.  Its bluetooth light comes on when I connect, but I've been unable to successfully transfer any data to it.  I'm not really sure what I should be using for the service and characteristic...

```
  sendData(device) {
    ble.isConnected(device.id, () => console.log('Still connected'), () => console.log('Not connected'));
    let data = new Uint8Array(1);
    data[0] = Utils.randomInt(1, 255);
    console.log(data);
    ble.write(device.id, 'FF10', 'FF11', data.buffer, () => console.log('data success'), () => console.log('data failure'));
    console.log('written');

    this.timeout = window.setTimeout(() => this.sendData(device), 10000);
  },
```

My output looks like this:
```
Still connected
data failure
28
written
Still connected
179
written
Still connected
107
written
...
```

Why wouldn't either of the callbacks be getting called every time?

don, 2016-02-25:
You should see the success callback called every time it writes. Maybe it's an ES6 problem? Have you tried with ES5 style functions?

azoerb, 2016-02-25:
I wouldn't think it would be, but I'll give that a shot and let you know.

azoerb, 2016-03-15:
Hey @don,

Switching over to ES5 style functions had no effect.

Digging around a bit, I found this message
```"155 F09 BLE1622351545 sAttempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference``` which was returned from the call to ```nativeApiProvider.get().exec(bridgeSecret, service, action, callbackId, argsJson);``` of line 938 of cordova.js.

Is there something wrong in my setup, or do you know what might be causing that message?

don, 2016-03-19:
It think you're passing bad UUIDs to `ble.write`. If you're using the Adafruit Bluefruit Feather board with the default UART service - use Service 6E400001-B5A3-F393-­E0A9-­E50E24DCCA9E and Characteristic 6E400002-B5A3-F393-­E0A9-­E50E24DCCA9E. See https://learn.adafruit.com/introducing-adafruit-ble-bluetooth-low-energy-friend/uart-service

FWIW you shouldn't need to check if you are connected every time. If the device disconnects, the failure callback from `ble.connect` will be called.

azoerb, 2016-03-20:
Awesome!  Thanks so much!  Somehow there were some weird characters in those UUIDs when I copied and pasted them (took me a bit to figure that one out), but after I removed them it worked like a charm.

Thanks for all of your help :)

closed 2016-03-20 by azoerb