don/cordova-plugin-ble-central #186: Unable to scan ble devices

Aman18493, 2016-02-23:
I am unable to scan other mobile devices from my android device.What can be the issue?

 ble.scan([], 50, app.onDiscoverDevice, app.onError);

 onDiscoverDevice: function(device) {

            console.log(JSON.stringify(device));
            console.log("Device Info");
            var listItem = document.createElement('li'),
                html = '<b>' + device.name + '</b><br/>' +
                    'RSSI: ' + device.rssi + '&nbsp;|&nbsp;' +
                    device.id;

            listItem.dataset.deviceId = device.id;  // TODO
            listItem.innerHTML = html;
            deviceList.appendChild(listItem);            
 }

I am scanning for devices and adding it to the list.

don, 2016-02-25:
At first glance the code looks OK.

Are there devices near you that are advertising Bluetooth Low Energy services? If not, you won't see anything. Use [nRF Master Control Panel](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to verify there are other BLE devices nearby.

If you don't have any Bluetooth LE peripherals, you can use [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8) to create virtual peripherals on iOS devices or nRF Master Control Panel to create virtual peripherals on newer Android devices.

[bleno](https://github.com/sandeepmistry/bleno) can be used to create Bluetooth Low Energy peripherals with Node.js on OS X, Linux and Windows.

don, 2016-02-25:
Duplicate of #173

closed 2016-02-25 by don