don/cordova-plugin-ble-central #187: How to get beacon data from Advertising data

MuraliKrishnaDev, 2016-02-23:
Hi,

I need to get Becaon data(UUID+Major+Minor+txpower) from ble.scan i.e. from advertising data,
Please suggest me on this,

Thanks and Regards,
Murali

don, 2016-02-25:
You can manually parse beacon data from the advertising info on Android, but I'd recommend using an iBeacon specific plugin like https://github.com/petermetz/cordova-plugin-ibeacon

MuraliKrishnaDev, 2016-02-25:
Hi,
Thanks for your response,
I need to get beacon data from Bluetooth scan i.e. UUID+Major+Minor+Txpower,
 https://github.com/petermetz/cordova-plugin-ibeacon this plugin is to scan the Ibeacon devices by providing on UUID ,Major ,Minor values  from user,
Please help me in this regard,
Best Regards,Murali

Date: Wed, 24 Feb 2016 19:48:10 -0800

don, 2016-03-19:
On Android, you can manually parse the iBeacon information from the advertising info. See #192 for info on how to get to the manufacturer data. Use the iBeacon spec for parsing the manufacturer data.

On iOS you can't do what you want. Apple forces you to use CoreLocation for beacons and removes the beacon information from the Bluetooth LE scan data.

closed 2016-03-19 by don